RaspbianFrance/raspisms #8: New system for MVC pingback

dam09fr, 2015-04-27:
As discussed on Raspbian-France website, I think the pingback function of the MVC framework is not the good way to do.

As I said, the IP collect is not the problem but you collect the full URL of RaspiSMS deployed by users and this database is accessible by everybody. Imagine if somebody discover a security bug he just have to collect this URL and launch a massive attack...

Another very important point : users who didn't read the source code did not know it! You MUST allow users to know that you are going to collect this data and you must show him this data before you send it.
Example: on first access to RaspiSMS, display a text explaining why you collect this data, what you want to do with, show him the exact content on data to transfer and get his aggrement to do it.
At least, please add an information to the documentation and on the download pages of the tool !

RaspbianFrance, 2015-04-27:
L'équipe de Raspbian France est relativement d'accord avec cette remarque.
Nous travaillerons à améliorer cela sur la prochaine version.
Le fait de collecter l'URL complète reste pour nous nécessaire, mais nous souhaitons rendre la démarche plus claire et compréhensible pour l'utilisateur.

Par ailleurs, le système est loin de nous satisfaire dans sa forme actuelle et retourne des résultats passablements mauvais et ne correspondant pas à la réalité.

Nous pensons retravailler ce système dans le sens d'une mise en avant impliquant plus l'utilisateur.

RaspbianFrance, 2015-08-15:
Le système de pingback a été supprimé !

closed 2015-08-15 by RaspbianFrance

memento, 2015-08-16:
Par quoi a-t-elle été remplacée ?

RaspbianFrance, 2015-08-16:
Le système n'a pas été remplacé.
Nous laissons tomber les statistiques et nous enregistrons simplement le téléchargement de l'application.
C'est moins précis mais c'est plus dans l'esprit du libre !

memento, 2015-08-16:
OK, c'est tout à votre honneur !
Bonne soirée !