RaspbianFrance/raspisms #14: Développement de future version

dam09fr, 2015-07-22:
Bonjour,

Pour des besoins propres à ma société, je vais prochainement reprendre une partie du code de RaspiSMS afin d'intégrer des fonctionalités dont nous avons besoin :
 - package Debian en respectant les best practices (localisation des dossiers, cron, config)
 - transfert des SMS entrants vers une ou plusieurs boites mails
 - limitation du nombre de SMS envoyés sur une période définie (minutes, heures, jours, semaine, mois) avec alerte mail lorsque le quota atteint 80% et lorsqu'il est dépassé
 - gestion des échecs d'envoi (méthode de failback sur API SMS de OVH)
 - alias /RaspiSMS/ personnalisable

Je pense que certaines de ces fonctions sont peut-être prévues dans la V2 dont vous parlez depuis un petit moment déjà et plutôt que de faire le travail deux fois (ou maintenir ma version "personnelle"), je vous propose de mettre à disposition de la communauté les sources de la version en cours de développement avec la roadmap des fonctionnalités, nous pourrions ainsi participer au développement de ce super outil.

Merci d'avance.

RaspbianFrance, 2015-07-22:
Bonjour,
certaines de ces fonctionnalités sont en effet prévues dans la version 2 de
RaspiSMS et nous serions ravis de les intégrer.
Quand penses-tu commencer à travailler dessus, que nous adaptions notre
emploie du temps !

:

dam09fr, 2015-07-22:
Je n'ai pas de planning ou de contraintes en particulier ...
Dès que vous serez prêts je le serai aussi !

dam09fr, 2015-08-16:
Hello !
Je suis en congés cette semaine, pensez-vous qu'on pourrait commencer les développements sur la future version d'ici une à deux semaines ?

RaspbianFrance, 2015-08-17:
Salut,
oui sans problèmes !
Pour tout dire nous sommes déjà entrain de mettre à jour RaspiSMS vers la
version 2.0 !
Par exemple, hier nous avons ajouter une interface permettant de discuter
plus facilement avec une personne !
Nous avons également revu l'organisation des fichiers pour rendre le tout
plus lisible, etc.

Nous sommes également en pleine écriture de la documentation du framework
Descartes, lequel sert de base à RaspiSMS. Une fois celle-ci terminée elle
viendra remplacer de façon plus adaptée une partie de la documentation de
RaspiSMS.

memento, 2015-08-19:
Salut,

Génial !
Dès que j'ai un peu de temps, je me lance dans les modifs pour la gestion des accusés de réception.

closed 2015-10-12 by RaspbianFrance