RaspbianFrance/raspisms #16: Timing des SMS peu précis

memento, 2015-07-22:
Bonjour Messieurs,

Il y a un petit soucis avec le timing des SMS. Ceux-ci sont associés à une minute près ce que je comprends puisqu'on fais les vérifications nécessaire à chaque minutes avant d'envoyer ou de recevoir une fournée de sms néanmoins les sms ne sont pas forcément affichés dans le bon ordre. C'est problématique.

Je pense qu'à la réception, une retouche du fichier parseSMS.sh serait nécessaire pour extraire le timing précis du sms.
À l'émission, il faudrait faire suivre le timing exact au clic du bouton (Enregistrer le SMS) pour ensuite l'injecter dans la commande qui fait appel à Gammu.

Qu'en pensez-vous ?

Bien à vous,
Alexandre.

RaspbianFrance, 2015-07-22:
Salut,
c'est une bonne idée. Le problème ne se posais pas pour l'instant puisque
le timing n'était pas très important mais ça serait à rajouter.
Il faudra vérifier dans ce cas sur l'envoie du SMS que les SMS sont bien
sélectionnés dans l'ordre d’envoi et pas par l'ordre de la première colonne.

memento, 2015-07-22:
Exact, ça pourrait faire l'affaire !

closed 2015-09-28 by RaspbianFrance