RaspbianFrance/raspisms #17: Bug : Problème d'accents

memento, 2015-07-27:
Bonjour, 
les accents ne passent pas en récéption. Ils sont remplacés pr des "?".

Dans la capture d'écran, le texte envoyé était : "Bon écolier à la ferme"
[image]
Peut-être que cela vient du codage caractère. Avez-vous une idée ?
C'est problématique.

Excellente journée !

memento, 2015-07-27:
En forçant l'UTF-8 dans /etc/gammurc et en redémararnt, rien ne change.

```
; Configuration file generated by gammu-detect.
; Please check The Gammu Manual for more information.

[gammu]
device = /dev/ttyUSB0
name = Phone on USB serial port HUAWEI_Technologies HUAWEI_Mobile
connection = at
gammucoding = utf8

[gammu1]
device = /dev/ttyUSB1
name = Phone on USB serial port HUAWEI_Technologies HUAWEI_Mobile
connection = at
gammucoding = utf8
```

memento, 2015-07-27:
J'ai essayé avec ce qui suit dans /etc/gammurc :

```
; Configuration file generated by gammu-detect.
; Please check The Gammu Manual for more information.

[gammu]
device = /dev/ttyUSB0
name = Phone on USB serial port HUAWEI_Technologies HUAWEI_Mobile
connection = at
gammucoding = iso88591

[gammu1]
device = /dev/ttyUSB1
name = Phone on USB serial port HUAWEI_Technologies HUAWEI_Mobile
connection = at
gammucoding = iso88591

```

ça commence par marcher après un redémarrage du service gammu-smsd ou un redémarrage du PI.
ça marche pour 5 ou 6 sms et après les accents sont de nouveaux transformés en "?".
Pourtant le fichier de conf n'a pas été changé à nouveau.

C'est de la magie noir.
Un retour d'expérience ?
Avez-vous eu le même problème ?

dam09fr, 2015-07-27:
J'ai le même problème.
Le problème vient bien de Gammu, les accents ne sont pas présents dans le fichier qui est déposé dans le dossier "receiveds".
L'encodage devrait être utf8 mais tous les caractères spéciaux se sont transformés en '?'.
Ce problème n'était pas présent auparavant, il est apparu à un moment donné (à identifier).

dam09fr, 2015-07-27:
J'ai modifié mon fichier de configuration : 
```
[gammu]
device = /dev/ttyUSB0
name = GSM_Dongle
connection = at
GammuCoding = utf8

[smsd]
(...)
```

+ Redémarrage du service gammu-smsd.
Et là ça marche, même au bout de plusieurs SMS.

RaspbianFrance, 2015-07-27:
Bonjour,
Le problème a été fixé par un membre de la communauté, il faut rajouter à
la section [gammu2] de /etc/gammu-smsdrc (c'est ce device qui est utilisé
par les SMS) la ligne :
GammuLoc = fr_FR.UTF8
et relancer gammu-smsd
et tout rentre dans l'ordre.

Le fixe corrige t-il le problème ?

dam09fr, 2015-07-27:
Chez moi l'ajout de la ligne : "GammuCoding = utf8" résout le problème.
L'ajout de "GammuLoc = fr_FR.UTF8" seule règle également le problème/
Du coup j'ai laissé les deux !

memento, 2015-07-27:
Bonjour,
J'ai fini par trouvé l'autre fichier de conf /etc/gammu-smsdrc et puis j'ai vu votre fix.
Je viens de tester et ça marche.
Je vais redémarrer le pi et envoyer/recevoir un dizaine de SMS pour être sûr.
Merci de votre réactivité en tout cas !

memento, 2015-07-27:
à l'émission, je n'ai plus de problème (le téléphone du destinataire reçoit tous les accents).
à la récéption, j'ai encore le problème 1 fois sur 3.

En plus du GammuLoc = fr_FR.UTF8 dans le fichier /etc/gammu-smsdrc, je devrais peut-être remettre le gammucoding = utf8 dans le fichier /etc/gammurc

Pouvez-vous me faire un copier/coller de votre fichier /etc/gammurc pour me donner une idée ?
Normalement sans gammucoding précisé, gammu doit prendre le codage lié au code caractère utilisé en local (locale) par débian, mais il semble que ça bloque.

Bien à vous,
Alexandre.

memento, 2015-07-27:
Correction, en regardant bien l'heure des messages, en récéption, 100 % des messages ont les caractères avec accent tarnsformés en "?".

Voici le contenu de mon /etc/gammurc

```
; Configuration file generated by gammu-detect.
; Please check The Gammu Manual for more information.

[gammu]
device = /dev/ttyUSB0
name = Phone on USB serial port HUAWEI_Technologies HUAWEI_Mobile
connection = at
gammucoding = utf8

[gammu1]
device = /dev/ttyUSB1
name = Phone on USB serial port HUAWEI_Technologies HUAWEI_Mobile
connection = at
gammucoding = utf8
```

memento, 2015-07-27:
Je précise qu'après chaque modification d'un fichier de conf, maintenant, je redémarre le Pi systématiquement pour ne prendre aucun risque (oui, je suis un barbare).

RaspbianFrance, 2015-07-27:
Pour l'instant, notre serveur RaspiSMS est débranché, nous ne pouvons pas
faire de test aujourd'hui.
Peut-être que dam09fr pourrais te donner la sienne.

memento, 2015-07-27:
J'ai remplacé gammucoding = utf8 par GammuCoding = utf8 dans /etc/gammurc au cas où gammu serait sensible à la casse. ça marche.
Je précise qu'en plus, j'ai dû faire un 
```
service gammu-smsd restart
```

en plus d'avoir redémarrer pour que la modification soit prise en compte. Etrange mais le problème semble résolu donc, je ne me plains pas.

Bien à vous !

dam09fr, 2015-07-27:
Hello,

Je n'ai rien touché dans le fichier "/etc/gammurc", son contenu est : 
```
[gammu]
device = /dev/ttyUSB0
name = GSM_Dongle
connection = at
```

Quant à mon fichier "/etc/gammu-smsdrc", il est configuré ainsi :
```
[gammu]
device = /dev/ttyUSB0
name = GSM_Dongle
connection = at
GammuCoding = utf8
GammuLoc = fr_FR.UTF8

[smsd]
pin = 0000
service = files
logfile = syslog
debuglevel = 0
RunOnReceive = /var/www/RaspiSMS/parseSMS.sh
inboxpath = /var/spool/gammu/inbox/
outboxpath = /var/spool/gammu/outbox/
sentsmspath = /var/spool/gammu/sent/
errorsmspath = /var/spool/gammu/error/
```

Important, à chaque modification de la configuration il faut redémarrer le service pour que les changements soient pris en compte, avec la commande suivante :
```
service gammu-smsd stop && sleep 5 && service gammu-smsd start
```
(la commande "restart" ne fonctionne pas très bien chez moi, elle tente de lancer le service alors que l'arrêt n'est pas finalisé)

memento, 2015-07-27:
Merci beaucoup, je vais tester ça !

memento, 2015-09-24:
Etrangement, de temps en temps, j'ai de nouveau ce problème d'accent quand mon RPI reçoit des SMS.
Le fichier de configuration n'a pourtant pas changé et un simple :
```
`service gammu-smsd stop && sleep 5 && service gammu-smsd start
```

résout le problème.

La vérité est ailleurs.

[image]

RaspbianFrance, 2015-09-24:
As-tu eu l'occasion de rencontrer le problème, puis de renvoyer exactement le même message une fois le service redémarrer, histoire de voir si le problème est bien résolu par le redémarrage ?

memento, 2015-09-24:
Oui oui, c'est ce que j'ai fait.

RaspbianFrance, 2015-09-24:
Bon, dans ce cas c'est un problème gammu, pas RaspiSMS :)
À eux de le gérer :b

closed 2015-09-24 by RaspbianFrance