RaspbianFrance/raspisms #19: Envoi et réception impossible après mise à jour du code

memento, 2015-07-31:
Bonjour,

Après avoir mis le code de RaspiSMS à jour (avant de faire des modifications), l'envoi et la réception snont impossible.
Le pod ne reçoit rien et les sms composés via l'interface de RaspiSMS ne partent jamais et restent dans scheduleds.

J'ai supprimé le contenu du dossier /var/www/RaspiSMS avant d'uploader le code RaspbianFrance/RaspiSMS

- Je précise que j'ai bien changé le propriétaire de tout le répertoire /var/www/RaspiSMS
```
chown www-data:www-data -R /var/www/RaspiSMS
```
- J'ai aussi mis tout le contenu du dossier de mon répertoire /var/www/RaspiSMS en 755
```
chmod 755 -R /var/www/RaspiSMS
```
(je suis sur un serveur de test qui n'est pas ouvert au WAN, juste le LAN)

- J'ai également recréé le dossier /var/www/RaspiSMS/receiveds

- J'ai remis mes informations de connexion à la bdd dans le fichier /var/www/RaspiSMS/mvc/conn_bdd.php

ça doit être bête mais quelque chose m'échape (?)

Note : Si vous m'aidez, je veux bien ajouter une page au wiki pour donner la marche à suivre à ceux qui veulent mettre à jour le code présent dans /var/www/RaspiSMS.

Bien à vous,

memento, 2015-07-31:
Note : Je vais tenter de chercher la "tâche" chargée de vérifier la présence de SMS prévus à l'envoi et stockés dans la table scheduleds de la base de donnée. Si quelqu'un peut m'en dire plus entre temps, ça m'arrangerait bien sûr (ce n'est aps dans la doc).

memento, 2015-07-31:
J'avais installé RaspiSMS avec le paquet .deb
J'ai regardé la [doc](http://raspbian-france.fr/envoyer-sms-raspberry-pi/) ( installation en manuel ) et j'ai eu quelques explications comme le fait de devoir donner la propriété du dossier /var/www/RaspiSMS/receiveds à l'utilisateur gammu.

Je sais aussi maintenant que c'est la crontab de l'utilisateur root qui doit être configuré avec l'ajout des lignes suivantes : 

> * * * * * cd /var/www/RaspiSMS && /var/www/RaspiSMS/console.php -c parseReceivedSMS
> * * * * * cd /var/www/RaspiSMS && /var/www/RaspiSMS/console.php -c sendScheduleds

Par contre, mon installation ne marche définitivement pas.
Si vous avez une idée..

Bien à vous,
Alexandre.

memento, 2015-07-31:
Dernière question, peut-on vraiment se contenter de copier le contenu du repository de github (RaspiSMS) dans /var/www ?
Je me pose la question. Si ça se trouve, ça marcherait chez moi si j'avais juste "mergé"/téléchargé le contenu du repositiry github par dessus le contenu de mon vieux dossier /var/www/RaspiSMS

Un conseil ?
C'est vrament dommage de ne pas guider le quidam dans la mise à jour de son RaspiSMS dans la doc.

RaspbianFrance, 2015-08-01:
Salut,
désolé du temps de  réponse mais nous avons beaucoup de projets en cours.

Pourrais-tu vérifier si le système gammu-smsd envoie correctement les messages, ceci directement en ligne de commande :
```bash
gammu-smsd-inject TEXT 'le numero cible' -text 'le texte du sms'; 
```

memento, 2015-08-01:
Salut,
Pas de souci, on est tous très occupés. ;)

Je ne pourrais pas me connecter au raspberry pi avant Lundi, malheureusement.
Je ferai une batterie de tests en posterai les résultats dès  Lundi.
Merci.

Bon week-end !

memento, 2015-08-03:
Bonjour,

Soit 06XXXXXXXX le numéro du destinataire :
```
root@rpi:/var/www/RaspiSMS# gammu-smsd-inject TEXT '06XXXXXXXX' -text 'le texte du sms'
gammu-smsd-inject[27919]: Created outbox message OUTC20150703_093113_00_06XXXXXXXX_sms0.smsbackup
Written message with ID /var/spool/gammu/outbox/OUTC20150703_093113_00_06XXXXXXXX_sms0.smsbackup
```

ça marche.
Le message est bien envoyé.

memento, 2015-08-03:
J'avance. Je veux ce qui va suivre le plus clair et précis possible pour que l'on puisse plus facilement me dépanner. Ne vous étonnez pas si ça traîne en longueur.

Je suis allé éditer ma crontab

```
crontab -e
```

Elle contenait (comme prévu dans la doc) :
```
* * * * * cd /var/www/RaspiSMS && /var/www/RaspiSMS/console.php -c parseReceivedSMS
* * * * * cd /var/www/RaspiSMS && /var/www/RaspiSMS/console.php -c sendScheduleds
```

J'ai testé la 1ère ligne qui doit parser les SMS reçus.
Elle doit, plus exactement se positionner dans le dosseir /var/www/RaspiSMS puis executer console.php avec le paramètre indiquant qu'il s'agit de parser les SMS reçus.

```
cd /var/www/RaspiSMS && /var/www/RaspiSMS/console.php -c parseReceivedSMS
```

Et là, c'est intéressant. Le shell me renvoie :

memento, 2015-08-03:
J'avance toujours.
J'ai fait une recherche des caractères de retour-chariot DOS/Windows sur tout le dossier /var/www/RaspiSMS : 

```
root@rpi:/var/www/RaspiSMS# grep -IUrl --color "^M"
js/plugins/flot/jquery.flot.pie.js
```

Sachez donc, pour votre information qu'un fichier "jquery.flot.pie.js" possède un de ces caractères diaboliques. Si vous avez un problème javascript un jour, voilà une piste !

Un autre point a attiré mon attention.

Quand je lance la commande suivante, on le sait, j'ai un problème :

```
root@rpi:/var/www/RaspiSMS# cd /var/www/RaspiSMS && /var/www/RaspiSMS/console.php -c parseReceivedSMS
-bash: /var/www/RaspiSMS/console.php : /usr/bin/php^M : mauvais interpréteur: Aucun fichier ou dossier de ce type
```
Dans la commande ci-dessus, ce n'est pas moi qui précise que pour executer le script console.php, il faut lancer le binaire /usr/bin/php. Le shell va chercher cette information quelque part. Il est possible que ce soit dans ce "quelque part" que l'on retrouve le retour charriot windowsien !

Pour étayer ma thèse, sachez que quand je lance la commande :

```
root@rpi:/var/www/RaspiSMS# cd /var/www/RaspiSMS && php /var/www/RaspiSMS/console.php -c parseReceivedSMS
```
Là, je n'ai pas de problème !

Je dois donc rechercher quel est, par exemple, le fichier de conf qui indique que out script php doit être lancé avec le binaire /usr/bin/php
J'ai le sentiment que c'est dans ce fichier de conf qu'on a le fameux "^M" de trop !

memento, 2015-08-03:
D'accord, je suis allé trop loin dans les hypothèses.
La ligne indiquant où se trouve le binaire php est tout simplement en première ligne du fichier console.php

```php
#!/usr/bin/php
<?php
        /**
         *      Cette page gère les scripts appelés en ligne de commande
         */

        ##############
        # INCLUSIONS #
        ##############
        //On va inclure l'ensemble des fichiers necessaires
        require_once('./mvc/constants.php');
        require_once('./mvc/autoload.php');
        require_once('./mvc/conn_bdd.php');
        
```

memento, 2015-08-03:
Effectivement, un simple : 
```
dos2unix console.php
```
dans /var/www/RaspiSMS a réglé le problème pour l'envoi.

Je peux maintenant executer la commande 

J'ai cependant une question car la réception ne fonctionne pas.
J'ai fait énormément de test de récéption et je me demande quel est le comportement de gammu quand la boîte de réception est saturée.

Vous avez une réponse ?

RaspbianFrance, 2015-08-03:
Salut,
merci de tes retours.
Je ne sais pas trop d'où peux sortir ce retour chariot à la noix puisque RaspiSMS n'a pas été développé sur Windows :/
Le plus étrange est que le problème apparaisse sur une commande et pas sur l'autre...

Je vais tout de suite faire passer dos2unix sur le projet et mettre à jour.

Je ne connais pas le comportement de gammu sur une bite de réception saturée. Je ne sais même pas vraiment si il est possible de saturer une boite de réception.

Les droits sur les fichiers sont-ils tous bon ? Les droits en écriture sur le répertoire "receiveds" ?

memento, 2015-08-03:
Droits sur le répertoire receiveds :
```
drwxr-xr-x  2 gammu    gammu    4096 juil. 31 14:54 receiveds
```
C'est bon.

Pour ce qui est de la boîte de réception, je parle bien sûr de la carte SIM. Oui, sur mon portable, quand ma carte SIM est pleine, je ne peux plus rien recevoir.

RaspbianFrance, 2015-08-03:
Au niveau de la carte SIM je n'y avais pas pensé.
Je ne sais pas du tout comment gammu gère cela. Il faudrait essayer de la vider depuis un téléphone pour voir.

memento, 2015-08-03:
J'ai vidé la carte SIM comme ci-dessous :

Tout d'abord il faut déterminer le numéro du "folder" (dossier) que l'on veut vider.
Pour avoir ces informations, il faut faire un :

```
root@rpi:/var/www/RaspiSMS# gammu --getsmsfolders
1. "          Boîte de réception", SIM memory, Inbox folder
2. "                Boîte d'envoi", SIM memory, Outbox folder
3. "          Boîte de réception", phone memory, Inbox folder
4. "                Boîte d'envoi", phone memory, Outbox folder
```

Dans notre cas, c'est le folder ___1___ qui nous intéresse.

```
root@rpi:/var/www/RaspiSMS# gammu --deleteallsms 1
Deleting SMS from "Boîte de réception" folder:
```

Par contre, RaspiSMS ne reçoit toujours rien.

J'ai une question. Souvent dans le code php il y a des instructions du type :
```php
$this->wlog(...);
```

Peux-tu me dire où sont logués les évennement ?
Dans quel fichier ? Peut-on consulter ce fameux log (j'aurais dû commencer par là, en fait).
Le code de internalController.php appelé par console.php est truffé de wlog(). Du coup, ça devrait me faire avancer.

memento, 2015-08-03:
Pour les logs :

Je vois qu'on activé les logs dans mvc/constants.php

```php
	//Réglages des logs
	define('LOG_ACTIVATED', 1); //On active les logs
```

Dans mvc/Controller.php : 
```php
protected function wlog($log)
		{
			if(!LOG_ACTIVATED) //Si on a désactivé les logs
			{
				return false; //On retourne faux
			}

			//On forge le message à logger
			$message = 'FROM [Controller : ' . $this->id . ' - ' . get_called_class() . '] - ' . str_replace("\0", "", $log);

			if(error_log($message)) //On log, si succes
			{
				return true; //On retourne vrai
			}

			return false;
		}
```

La fonction error_log($message) "envoie un message d'erreur à l'historique du serveur web ou un fichier".

D'après cette (doc)[http://php.net/manual/fr/function.error-log.php] : 
> error_log
> 
> (PHP 4, PHP 5, PHP 7)
> error_log — Envoi un message d'erreur vers le gestionnaire d'erreurs défini
> 
> Description ¶
> 
> bool error_log ( string $message [, int $message_type = 0 [, string $destination [, string 
> $extra_headers ]]] )
> Envoi un message d'erreur à l'historique d'erreur du serveur web ou à un fichier.

Or, il n'y a rien dans /var/log/apache/

RaspbianFrance, 2015-08-03:
Normalement le fichier de log est renseigné dans ta configuration Apache.

memento, 2015-08-03:
J'ai fait plusieurs essais pour les logs dans les 2 php.ini que j'ai mais rien de nouveau.
Il n'y a pas d'erreur php.

Quand je fais la commande suivante après m'être envoyé plusieurs sms, je constate que je n'ai aucun SMS
```
root@rpi:~# gammu getallsms

0 SMS parts in 0 SMS sequences
```

On dirait que le problème vient de gammu

RaspbianFrance, 2015-08-03:
Le problème viens donc à priori de la réception par le dongle GSM. Là dessus nous n'allons hélas pas trop pouvoir t'aider :/

Pour les logs la configuration est dans Apache, probablement dans un fichier du type "/etc/apache2/sites-available/000-default.conf"

memento, 2015-08-03:
___Je reviens sur ce que j'ai dit___

Gammu a bien réagi à la récéption du SMS, par contre ce dernier à rencontré une erreur. Dans /var/log/syslog :
```
Aug  3 15:42:23 rpi gammu-smsd[2674]: Received message from: +336XXXXXXXX
Aug  3 15:42:23 rpi gammu-smsd[2674]: Read 1 messages
Aug  3 15:42:23 rpi gammu-smsd[2674]: Received IN20150803_154219_00_+336XXXXXXXX_00.txt
Aug  3 15:42:23 rpi gammu-smsd[3347]: Starting run on receive: /var/www/RaspiSMS/parseSMS.sh IN20150803_154219_00_+336XXXXXXXX_00.txt
Aug  3 15:42:23 rpi gammu-smsd[2674]: Process failed with exit status 127
```

Je vais voir ce que c'est que cette erreur.

memento, 2015-08-03:
___J'ai trouvé.___

```
Aug  3 15:42:23 rpi gammu-smsd[2674]: Process failed with exit status 127
```
Celà signifie que parseSMS.sh ne s'est pas executé normalement.
En executant parseSMS.sh, devinez ce que j'ai trouvé en sortie console ?

```
root@rpi:/var/www/RaspiSMS# ./parseSMS.sh
-bash: ./parseSMS.sh : /bin/sh^M : mauvais interpréteur: Aucun fichier ou dossier de ce type
```

Encore un caractère retour charriot windows de m****

Du coup, encore une fois, l'utilitaire dos2unix va nous néttoyer tout ça :
```
root@rpi:/var/www/RaspiSMS# dos2unix parseSMS.sh
dos2unix: converting file parseSMS.sh to Unix format ...
```

Résolu.
Il ne faudra pas oublier de repasser tout le projet au dos2unix car je suis formel, je n'ai édité le fichier parseSMS.sh sous windows. J'ai directement cloné le répertoire Github.

closed 2015-08-03 by memento

memento, 2015-09-21:
Bonjour,
J'ai encore importé le code aujourd’hui et j'ai encore eu le fameux retour charriot.

RaspbianFrance, 2015-09-21:
Ah, pourtant le projet est configurer pour régler le problème comme un
grand --'.
Je crois qu'il va falloir que tu corriges manuellement, du moins pour
l'instant.

memento, 2015-09-22:
Oui, ne t'inquiète pas, j'ai déjà repassé console.php et parseSMS.sh à la moulinette (dos2unix).