RaspbianFrance/raspisms #22: ... (3 petiis points) devant une variable PHP ? C'est normal ?

memento, 2015-09-21:
Bonjour,

J'ai pris la dernière version.
Quand j'essaye d'afficher des pages comme :
SMS => http://mon-site.fr/RaspiSMS/scheduleds/
>_ Commandes => http://mon-site.fr/RaspiSMS/commands/
Répertoire>Contacts => http://mon-site.fr/contacts
Répertoire>Groupes => http://mon-site.fr/groups
Utilisateurs => http://mon-site.fr/RaspiSMS/users/

J'ai ne page blanche.
Le log me dit par exemple :

```
Sep 21 13:28:48 rpi apache2: PHP Parse error:  syntax error, unexpected '.', expecting '&' or variable (T_VARIABLE) in /var/www/RaspiSMS/controllers/commands.php on line 39
```
ou encore :

```
Sep 21 13:33:14 rpi apache2: PHP Parse error:  syntax error, unexpected '.', expecting '&' or variable (T_VARIABLE) in /var/www/RaspiSMS/controllers/scheduleds.php on line 37
```

PHP ne supporte pas les 3 petits points ...

Par exemple, dans le fichier XXX, j'ai :
```php
public function delete($csrf, ...$ids)
```

Je ne connaissais pas les ... devant une variable. :)

memento, 2015-09-21:
D'accord, l'opérateur ... est disponible à partir de la version 5.6 (et plus) de PHP
()
Sur mon raspbian je n'ai que la version 5.4.41

memento, 2015-09-21:
En faisant une mise à jour de PHP 
```
sudo apt-get update && sudo apt-get install php5
```

Je n'ai pu passer qu'à la version : 
```
PHP 5.4.45-0
```

je vais compiler les sources du paquet php5 mais pour les utilisateurs lambdas qui ne pourront/voudront que lancer l'installation des paquets avec ___apt-get___, l'utilisation de cet opérateur ... n'est à mon avis pas judicieux.

memento, 2015-09-21:
Ok, la mise à jour vers cette version de php obligera également à changer de version d'apache sachant que la mise à jour d'apache (2.2 vers 2.4) n'a pas la même façon de gérer les fichiers de conf, plus d'autres modifications de la configuration.
ça commence à faire beaucoup pour un opérateur, non ?

Quelqu'un a une astuce ? (un guru de php)

Bien à vous !

memento, 2015-09-21:
Le fait de supprimer les ... ne semble pas perturber les contrôleurs.
J'espère que ça aidera quelqu'un !

RaspbianFrance, 2015-09-21:
Salut, les "..." permettent de récupérer plusieurs variables.
Dans le cas présent cela est par exemple utile pour supprimer plusieurs contacts.
Je pense qu'il va effectivement falloir que nous trouvions une façon de gérer cela en restant dans une version < à 5.6 (même si c'est bien dommage).

Je pense qu'il devrait être possible d'utiliser les fonctions func_get_args(). 
C'est nettement moins élégant, mais plus rétro-compatible;
Je me penche sur la question dès maintenant.

RaspbianFrance, 2015-09-21:
Le problème viens d'être corrigé.
La solution choisie est l’utilisation de la fonctions func_get_args.

closed 2015-09-21 by RaspbianFrance

memento, 2015-09-22:
La modification est déjà dans le code ?

RaspbianFrance, 2015-09-22:
Oui, nous avons mis à jour tout ça :)

memento, 2015-09-22:
Merci !
Quel boulot vous faites !

RaspbianFrance, 2015-09-22:
Il pleut tout le temps en charente en ce moment, faut bien s'occuper !