RaspbianFrance/raspisms #23: Contacts : Il faudrait obliger le format +33 6xxxxxx

memento, 2015-09-22:
Bonjour, 

Concernant les contacts, il faudrait obliger l'utilisateur à entrer le numéro au format international à la création d'un contact car quand on reçoit un message en réponse, c'est toujours au format +336xxxxxxxx. Du coup, RaspiSMS l'identifie comme étant un numéro inconnu et la conversation est "cassée".

Bien à vous !

RaspbianFrance, 2015-09-22:
Salut,
je pense qu'il faut effectivement privilégier les numéros internationaux, mais il me semblerait dommage d'imposer ce choix aux utilisateurs.
D'autant que c'est une notation qui n'est vraiment pas naturelle.

Nous pourrions peut-être ajouter une petite ligne d'avertissement dans les formulaires contenant des numéros.

RaspbianFrance, 2015-09-22:
Nous venons d'ajouter le message précédemment évoqué sur l'envoi de SMS et la création de contacts !

memento, 2015-09-22:
En fait, c'est aussi gênant pour la gestion des accusées de réception car quand un accusé arrive, il est signé par un numéro au format international alors que le contact peut être enregistré avec un numéro au format local. Du coup, pas de correspondance possible entre le message précédemment envoyé et l'accusé de réception.

Je propose de couper la poire en deux.
Plutôt que d'imposer à l'utilisateur d'écrire un numéro en +336..., je propose le traitement suivant à l'ajout du contact :

Si le numéro est à 10 chiffre (pour être sûr) et qu'il commence par un "0", alors on transforme le numéro de la forme 0123456789 vers la forme +33123456789 juste avant l'ajout dans la base de données. ça peut être rajouté comme fonction dans la classes des utilitaires.

Remarque : C'est vrai que quand je suis dans la rue ( :) ) et que j'entre un nouveau contact dans mon répertoire, ça peut m'arranger de n'entrer que la forme locale par contre ça ne me dérange pas si le numéro est converti sans effort de ma part.

Qu'en pensez-vous ?

Bien à vous.

memento, 2015-09-22:
Remarque 2 : En plus, quand on entre des contacts, parfois on a le courage de taper le +336... et parfois non. Du coup, le répertoire n'est "beau" car la liste des numéros n'est pas régulière (oui, je suis un ___geek maniac___).

Ok, c'est un argument qui pèse pour la moitié d'un mais pour les maniaques du contrôle comme moi, ça compte et force est de constater que chez les utilisateurs de Raspberry Pi, il doit y en avoir beaucoup.

Bon allez, je n'insiste plus, j'attends votre réponse.

RaspbianFrance, 2015-09-22:
Nous avions déjà pensé à cette solution, mais elle n'est pas viable en pratique.
Le 0 correspond à +33 pour la France, mais il correspondra à une valeur complètement différente pour la Belgique par exemple.

À la limite, il serait envisageable d'ajouter une configuration supplémentaire qui permette d'activer l'auto-transformation des numéros, et une autre qui permette de définir la chaîne de remplacement.

RaspbianFrance, 2015-09-22:
Nous sommes entrain de finir le code des accusés de réception.
Au passage, nous mettons à jour la version de Descartes utilisée par RaspiSMS (nous en aurons besoin pour pouvoir gérer plus en détails les requêtes SQL, ce qui sera nécessaire pour gérer les accusés de réception de façon élégante).

memento, 2015-09-22:
Oui, effectivement, bien vu ! Le traitement devrait changer selon la nationalité et il n'y a pas que les nombre de chiffre composants le numéro égal à 10 qui rentre en jeu.

Pour les accusés, j'espère que mes indications dans les issues vous ont servis.
Vous pensez avoir terminé quand ?
Je continue tout de même de mon côté (pour le sport) ;)

Pour les accusés, je prends une logique de pile (first in, first out) et je ne peux pas être plus précis que ça étant donné que les accusés n'arrivent pas avec un timestamp ou un id du message précédemment envoyé. A moins que vous ayez trouvé un contournement ?

Vous servez-vous d'une fonctionnalité de Gammu, ou bien une pile (comme moi) ?

RaspbianFrance, 2015-09-22:
Nous utilisons également une pile (il ne semble hélas pas exister de méthode plus précise).
Nous aurons probablement fini d'ici une ou deux heures ;)

RaspbianFrance, 2015-09-22:
Les accusés de réception sont en place.
Le système de pile n'est pas parfait mais nous avons essayé d'améliorer le tout en ne prenant en compte que les SMS en attente de réception depuis moins de 24h !

memento, 2015-09-22:
Une ou deux heures ? Génial !

Pour la précision, il faut passer Gammu en mode base de données ([par exemple MySQL](http://wammu.eu/docs/manual/smsd/mysql.html)). Là, nous sommes en mode fichier. Il faut savoir qu'en mode base de données, lorsqu'un accusé arrive, la ligne de la table des SMS envoyés dans la base de données est automatiquement mise à jour avec le statut.

Ainsi, une simple requête vers la base de données de Gammu peut nous renseigner.
On pourrait ainsi implémenté le tout comme une tâche périodique visant à synchroniser les informations contenues dans les deux tables des bases de données.

Au passage, si on avait pris Gammu en mode base de données, il y aurait eu moins de boulot pour RaspiSMS en général car on n'aurait eu qu'à attaquer la base de données (pour tout un tas d'actions). Cependant, c'est le lot de tout projet mené avec engouement et il fallait bien choisir une voie.

Bravo à vous !

Pour RaspiSMS en parallèle de Gammu en mode MySQL, ça peut toujours faire l'objet d'une version 2.X, qui sait ! Comme à mon habitude, je posterai mes dernières pistes à ce sujet (entre autre). J'ai dû interrompre le développement de la gestion des accusés pendant quelques semaines vu la quantité de taf au bureau.

J'ai hâte de voir votre approche du problème des accusés.

Bonne soirée !

RaspbianFrance, 2015-09-22:
À la base le projet a été pensé pour gammu et pas gammu-smsd.
Par ailleurs, le mode base de données nous demanderais toujours de passer par des méthodes d'envoi utilisant les lignes de commande afin de remplir correctement l'ensemble des champs.

En revanche, il serait envisageable de créer un système permettant de faire le lien entre les deux bases de données afin de récupérer des informations plus précises, notamment sur l'état des SMS.

memento, 2015-09-22:
___Attention___ : sauf erreur de ma part, votre approche ne prend pas en compte qu'un accusé de réception n'est un lié qu'à une tranche de 160 caractères du message. Du coup, par exemple, pour un message de 165 caractères (2 accusés), suivit d'un message de 10 caractères (1 accusé) au même destinataire, vous recevrez 3 accusés au lieu de deux pour le même numéro. C'est la limite principale de l'approche par pile "aveugle" (sans id ou timestamp désignant le message précédemment envoyé).

Pour contourner ce problème, de mon côté, j'ai ajouté un champ "slices_to_confirm" (tranches à confirmer) qui est renseigné dès l'ajout du message à la table sendeds avec le nombre de sms qui composent le message (Cf. modulo et division entière de la longeur du message par l'entier 160).
Ensuite, à chaque traitement d'accusé concernant le destinataire, je décrémente le champ "slices_to_confirm" et quand il arrive à zéro, j'en profite pour passer le message en "Delivered". Mission accomplie. Je me suis même permis le luxe de gérer les accusés de réception "Failed" et plus particulièrement quand, dans le flot de sms de 160 caractères, il y en avait un qui foire ("Failed"). Résultat, je décrémente toujours le champ "slices_to_confirm" mais si par malheur j'ai un "Failed" dans le lot, je passe directement le message en "Failed" et ce dernier ne sera plus mis à jour.

On était au coude à coude, ce soir. ;)

reposez-vous bien !

RaspbianFrance, 2015-09-22:
Bien vu, nous avions totalement oublié le cas des SMS de plus de 160 caractères. Nous allons regarder ça !

RaspbianFrance, 2015-09-22:
Et voilà, c'est mis en place !

memento, 2015-09-23:
Fantastique !

memento, 2015-09-23:
Comme je l'ai dit plus tôt, le format local des numéros pose des problèmes. 

Le dernier en date :
L'acceptation du format local de numéro (0612345678) à la création du contact, ou encore à la création d'un message en utilisant le champ "Numéros cibles" fait échouer le système de gestion des accusés de réception. En d'autres termes, tout message envoyé à un numéro en 06 ne sera officiellement ___jamais délivré___. La raison est que le réseau GSM renverra ___toujours___ l'accusé de réception avec un format de numéro international.

Pour un message envoyé au 0612345678, je recevrai un accusé semblant venir du +33612345678. Cet accusé ira donc aux oubliettes.

Dans les deux formulaire, pourquoi ne pas proposer (à gauche du champ numéro) une petite combobox sympa avec un drapeau (mis par défaut sur la France) et/ou proposant l'indicatif du pays (par défaut +33) ?

La gestion des accusés de réception est une super valeur ajoutée pour le projet, ça serait dommage, même si n'est que mon avis.

Bien à vous !

memento, 2015-09-23:
J'ai vu que vous utilisiez JQuery.
Voilà un bon exemple pour le numéro de téléphone : [ici](http://www.jqueryscript.net/demo/jQuery-International-Telephone-Input-With-Flags-Dial-Codes/).

RaspbianFrance, 2015-09-23:
Ce système a l'air assez prometteur, je ferais peut-être quelques tests ce soir !
Le problème sera de faire cohabiter ce plugin avec celui permettant l'auto-complétion des groupes, etc.
Je verrais si j'arrive à faire un système assez propre !

memento, 2015-09-23:
Cool !

memento, 2015-09-24:
Dans templates/scheduleds/add.php, prennons le div du champs input du numéro de téléphone :

```html
<div class="form-group">
									<label>Numéros cibles <span class="italic small-text text-danger"> - Vous devriez utiliser un numéro international (Ex. : +33612345678)</span></label>
									<input class="add-numbers form-control" name="numbers[]"/>
								</div>
```

RaspbianFrance, 2015-09-28:
Nous venons d'ajouter l'auto-complétion des numéros avec préfixes internationaux.
Nous avons retenu la solution du plugin présenté plus haut !

Nous ajouterons un réglages permettant de choisir les langues préférées et la langue par défaut pour les numéros très rapidement !

RaspbianFrance, 2015-09-28:
Nous avons ajouté les réglages sur l'internationalisation des numéros (pays par défaut et pays préférés).
Nous avons également unifié les réglages.

closed 2015-09-28 by RaspbianFrance

memento, 2015-09-29:
Génial !

Nico80, 2016-01-10:
Bonjour, à tous, 
Petite question à propos du +33, comment le passer dans une URL.
J'ai essayé plusieurs possibilités mais rien ne marche...
http://192.168.1.35/RaspiSMS/smsAPI/?email=****************@yahoo.fr&password=*****&numbers=33623******&text=test54

http://192.168.1.35/RaspiSMS/smsAPI/?email=****************@yahoo.fr&password=*****&numbers=+33623******&text=test54
Rien ne marche...

seul le 0623****** fonctionne.
Si vous avez une idée.
Merci.

RaspbianFrance, 2016-01-10:
Salut,
je pense qu'il faut que tu convertisses le caractère "+" en son équivalent
au format url "%2B".

Nico80, 2016-01-10:
Merci bonne idée, c'est vrai j'ai pas pensé à ça...
Je vais tester.

memento, 2016-01-11:
Bonjour,

Merci pour le %2B à la place du +, a peut toujours servir !

Bonne journée !