RaspbianFrance/raspisms #24: Constant RASPISMS_SETTINGS_TRANSFER already defined

memento, 2015-09-23:
Bonjour,

Après mise à jour, les lignes suivantes apparaissent dans mon /var/log/syslog

```
Sep 23 11:52:02 rpi apache2: PHP Notice:  Constant RASPISMS_SETTINGS_TRANSFER already defined in /var/www/RaspiSMS/index.php on line 25
Sep 23 11:52:02 rpi apache2: PHP Notice:  Constant RASPISMS_SETTINGS_SMS_STOP already defined in /var/www/RaspiSMS/index.php on line 25
Sep 23 11:52:02 rpi apache2: PHP Notice:  Constant RASPISMS_SETTINGS_DETECT_URL already defined in /var/www/RaspiSMS/index.php on line 25
Sep 23 11:52:12 rpi apache2: PHP Notice:  Constant RASPISMS_SETTINGS_TRANSFER already defined in /var/www/RaspiSMS/index.php on line 25
Sep 23 11:52:12 rpi apache2: PHP Notice:  Constant RASPISMS_SETTINGS_SMS_STOP already defined in /var/www/RaspiSMS/index.php on line 25
Sep 23 11:52:12 rpi apache2: PHP Notice:  Constant RASPISMS_SETTINGS_DETECT_URL already defined in /var/www/RaspiSMS/index.php on line 25

```

Je ne parviens pas non plus à envoyer de sms (l'ajout se passe bien via l'interface ais le message n'est jamais envoyé et reste dans la pile scheduleds ). Je ne sais pas si c'est lié.

memento, 2015-09-23:
Au cas où, voilà le contenu de ma crontab :

```
* * * * * cd /var/www/RaspiSMS && /var/www/RaspiSMS/console.php -c parseReceivedSMS
* * * * * cd /var/www/RaspiSMS && /var/www/RaspiSMS/console.php -c sendScheduleds
```

memento, 2015-09-23:
Dans le fichier console.php, on a le bloc suivant :

```php
	//On va ajouter les réglages globaux de RaspiSMS modifiables via l'interface
	$settings = $db->getFromTableWhere('settings');
	foreach ($settings as $setting)
	{
		define('RASPISMS_SETTINGS_' . mb_convert_case($setting['name'],  MB_CASE_UPPER), $setting['value']);
	}
```

Et dans le fichier index.php : 
```php
	//On va ajouter les réglages globaux de RaspiSMS modifiables via l'interface
	$settings = $db->getFromTableWhere('settings');
	foreach ($settings as $setting)
	{
		define('RASPISMS_SETTINGS_' . mb_convert_case($setting['name'],  MB_CASE_UPPER), $setting['value']);
	}
```

memento, 2015-09-23:
Pour l'impossibilité d’envoyer et de recevoir, je n'y croyais aps mais il s'agit à nouveau des retour-charriot.
Étrangement, je n'ai pas eu les messages d'erreur d’antan.

J'ai passé comme d'habitude console.php et parseSMS.php au ___dos2unix___ et le problème d'envoi était résolu.

Reste le problème de réception.

J'ai une nouvelle erreur (exception) :

```
Sep 23 13:35:24 rpi php: FROM [Controller : 56028e65e982a - internalConsole] - Delivered or Failed SMS for +33677073278
Sep 23 13:35:24 rpi php: PHP Fatal error:  Uncaught exception 'Exception' with message 'DateInterval::__construct(): Unknown or bad format (P12H)' in /var/www/RaspiSMS/controllers/internalConsole.php:258#012Stack trace:#012#0 /var/www/RaspiSMS/controllers/internalConsole.php(258): DateInterval->__construct('P12H')#012#1 /var/www/RaspiSMS/console.php(54): internalConsole->parseReceivedSMS()#012#2 {main}#012  thrown in /var/www/RaspiSMS/controllers/internalConsole.php on line 258
```

memento, 2015-09-23:
Dans internalConsole.php, remplacer ___P12H___ par ___PT12H___ a réglé le problème :

```php
$interval = new DateInterval('PT12H');
```

RaspbianFrance, 2015-09-23:
Salut,
nous venons de voir le problème (réveil un peu tardif ;)).
Nous corrigeons le problème tout de suite.

memento, 2015-09-23:
Pas de soucis. J'ai fait une request pour la correction. ça marche chez moi depuis.

RaspbianFrance, 2015-09-23:
Pour le problème des constantes, je pense que tu as des lignes en double dans la table "settings".
Il est normal que le bloc apparaisse dans les deux fichiers, ceux-ci étant parfaitement indépendants.

memento, 2015-09-23:
Étant donné que j'ai refait ma base de données depuis, je ne peux pas vérifier mais ça doit être ça. ;)

closed 2015-09-23 by memento

RaspbianFrance, 2015-09-23:
Nous venons d'ajouter une contrainte unique sur le nom pour éviter ce genre de problèmes.