RaspbianFrance/raspisms #26: Documentation : mise à jour nécessaire

memento, 2015-09-23:
J'utilise également RaspiSMS avec ma propre interface en dur et je me suis rendu compte qu'avec la nouvelle implémentation de votre Framework Descartes, l'url complète à fournir ne convenait plus.

>#Requête POST
>curl -X POST http://URLCIBLE/smsAPI/send/

L'utilisateur doit maintenant enlever le ___send/___ et fournir une url complète du type :

>#Requête POST
>curl -X POST http://URLCIBLE/smsAPI/

Changement à effectuer [ici](http://raspisms.raspbian-france.fr/documentation/#api-parameters-examples)

RaspbianFrance, 2015-09-23:
Merci du retour, nous l'avions fait sans même nous en rendre compte !
La documentation complète méritera une mise à jour, mais celle-ci ne sera effectuée qu'une fois la version 2 finie et prête à être distribuée.
Par ailleurs, la documentation sera en partie réduite afin de renvoyer vers celle du framework Descartes, en cours d'écriture !

memento, 2015-12-29:
Un autre utilisateur a été victime de l'erreur de doc ;)
Statistiquement, d'autre utilisateurs doivent avoir eu le même problème.
Une petite correction ?

Bonne soirée et joyeuses fêtes !

memento, 2015-12-29:
Limite, je le fais si vous me donner un accès (même temporaire).

RaspbianFrance, 2015-12-30:
C'est réglé !
Il faut vraiment que nous refassions la documentation complète d'une façon qui nous satisfasse d'avantage.
Nous envisageons d'utiliser le Wiki de GitHub.
La difficulté, c'est que pour bien faire nous devons commencé par la documentation du framework Descartes. Par ailleurs, nous avons pas mal d'autres projets qui actuellement nous demande du temps et qui passent avant la documentation. Il faudra donc probablement attendre encore un peu !

closed 2015-12-30 by RaspbianFrance