RaspbianFrance/raspisms #27: Solution pour éviter les erreurs de non-correspondance entre scheduleds et scheduleds_[numbers|contacts|groups] en cas de mode multiutilisateurs

memento, 2015-09-24:
Bonjour,

Ceci est un problème au niveau de la démarche d'enregistrement d'un nouveau SMS programmé.

Voyons le code (controllers/scheduleds.php).
Première étape, on ajoute le sms (sans destinataire) dans la table ___scheduleds___ à partir de la ligne 166 :

```php
			if (!$db->insertIntoTable('scheduleds', ['at' => $date, 'content' => $content, 'progress' => false]))
			{
				if (!$api)
				{
					$_SESSION['errormessage'] = 'Impossible de créer ce SMS.';
					header('Location: ' . $this->generateUrl('scheduleds'));
				}
				return false;
			}
[...]
```
Puis, à la ligne 176, on récupère l'ide du dernier message ajouté à la table (scheduleds, encore une fois) :

```php
[...]

			$id_scheduled = $db->lastId();
[...]
```

Logique, puisqu'on devra préciser l'id au moment de rajouter une ou plusieurs lignes aux tables scheduleds_numbers et/ou scheduleds_contacts et/ou scheduleds_groups (ligne 187) :

```php
[...]
			foreach ($numbers as $number)
			{
				if (!$number = internalTools::parsePhone($number))
				{
					$errors = true;
					continue;
				}

				if (!$db->insertIntoTable('scheduleds_numbers', ['id_scheduled' => $id_scheduled, 'number' => $number]))
				{
					$errors = true;
				}
			}

			foreach ($contacts as $id_contact)
			{
				if (!$db->insertIntoTable('scheduleds_contacts', ['id_scheduled' => $id_scheduled, 'id_contact' => $id_contact]))
				{
					$errors = true;
				}
			}

			foreach ($groups as $id_group)
			{
				if (!$db->insertIntoTable('scheduleds_groups', ['id_scheduled' => $id_scheduled, 'id_group' => $id_group]))
				{
					$errors = true;
				}
			}

```

L'approche est correcte quand il n'y a qu'un seul utilisateur qui ajoute des SMS. Seulement voilà, RaspiSMS est prévu pour être multi-utilisateur. Par exemple, c'est moi, un de mes serveur peut avoir à envoyer des extraits de log par SMS au moment d'un crash ou de l'apparition d'un événement important. Je peux également naviguer sur l'interface RaspiSMS pour envoyer un rendez-vous par SMS à une liste d'amis. Enfin, ma femme peut également utiliser mon application client riche qui va utiliser des [requêtes POST](http://raspisms.raspbian-france.fr/documentation/#api-parameters-examples) pour envoyer des SMS via l'API RaspiSMS. Je suppose que vous voyer où je veux en venir, Entre l'ajout du SMS à la table scheduleds (depuis le WAN) et la récupération du dernier id (toujours depuis le WAN), il peut se passer assez de temps pour permettre à mon serveur d'ajouter une floppée de SMS à la table scheduleds (depuis le LAN).

Par conséquent, de mauvais SMS pourraient être envoyés à de mauvais destinataires. Imagnez si ma femme envoi des traces de crash serveur à ses copines.. le tout sans avoir la possibilité d'être averti par log d'erreur.

Heureusement, il existe une solution en MySQL (plusieurs solutions, en fait) pour ajouter un nouveau sms à la table scheduled tout en récupérant l'id généré pour l'occasion : [MySQL : Obtenir la valeur de l'auto-incrément](http://dev.mysql.com/doc/connector-odbc/en/connector-odbc-usagenotes-functionality-last-insert-id.html).

RaspbianFrance, 2015-09-24:
Salut,
merci de la remonté, mais pour cette fois il s'agit d'une fausse alerte ;)
Tout d'abord, le temps entre les deux événements est extrêmement faible (genre 0,01s) et utiliser MySQL présenterais exactement les mêmes inconvénients (il faudrait faire deux requêtes pour MySQL).

Ensuite, lastId() repose sur  la méthode PDO "lastInsertId". Laquelle récupère le dernier ID inséré **pour la connexion en cours** *(voir par exemple http://stackoverflow.com/questions/9891945/pdo-last-insert-id-always-the-right-one)*. Comme chaque appel créer sa propre connexion, il n'y a pas de risque de confusion :)

closed 2015-09-24 by RaspbianFrance

memento, 2015-09-24:
Cool, je ne savais pas pour lastId() !
;)