RaspbianFrance/raspisms #28: Oct 12 13:53:58 rpi php: PHP Fatal error:  Call to undefined method DataBase::getAll() in /var/www/RaspiSMS/controllers/internalConsole.php on line 330

memento, 2015-10-12:
Bonjour,

Quand j'envoie une commande à mon RaspiSMS, ça plante et mon syslog renvoie :
```
Oct 12 13:53:58 rpi php: PHP Fatal error:  Call to undefined method DataBase::getAll() in /var/www/RaspiSMS/controllers/internalConsole.php on line 330
```

Rappel de la ligne dans internalConsole.php (ligne 330 chez moi) :
 
```php
$commands = $db->getAll('commands');
```

Rappel du bloc de commandes en question :
```php
					//Si on reçu des identifiants
					if (array_key_exists('LOGIN', $flags) && array_key_exists('PASSWORD', $flags))
					{
						//Si on a bien un utilisateur avec les identifiants reçus
						$user = $db->getUserFromEmail($flags['LOGIN']);
						$this->wlog('We found ' . count($user) . ' users');
						if ($user && $user['password'] == sha1($flags['PASSWORD']))
						{
							$this->wlog('Password is valid');
							//On va passer en revue toutes les commandes, pour voir si on en trouve dans ce message
							$commands = $db->getAll('commands');

							$this->wlog('We found ' . count($commands) . ' commands');
							foreach ($commands as $command)
							{
								$command_name = mb_strtoupper($command['name']);
								if (array_key_exists($command_name, $flags))
								{
									$this->wlog('We found command ' . $command_name);
									
									//Si la commande ne nécessite pas d'être admin, ou si on est admin
									if (!$command['admin'] || $user['admin'])
									{
										$this->wlog('And the count is ok');
										$found_commands[$command_name] = PWD_SCRIPTS . $command['script'] . escapeshellcmd($flags[$command_name]);
									}
								}
							}
						}
					}
```

memento, 2015-10-12:
Quelqu'un a déjà testé les commandes envoyées par SMS ?

memento, 2015-10-12:
J'ai compris.
Dans la version 1 (la vieille version), j'ai regardé le code.
Il y a une fonction/méthode getAll() dans le fichier mvc/Model.php :

```php
		/**
		 * Cette fonction permet de récupérer une table complète, éventuellement en la triant par une colonne, éventuellement en limitant le nombre de résultat, ou en sautant certains (notamment pour de la pagination)
		 * @param string $table : Le nom de la table a récupérer
		 * @param string $order_by : Le nom de la colonne par laquelle on veux trier les résultats. Si non fourni, tri automatique
		 * @param string $desc : L'ordre de tri (asc ou desc). Si non défini, ordre par défaut (ASC)
		 * @param string $limit : Le nombre maximum de résultats à récupérer (par défaut pas le limite)
		 * @param string $offset : Le nombre de résultats à ignorer (par défaut pas de résultats ignorés)
		 * @return array : Tableau avec dans chaque case une ligne de la base
		 */
		public function getAll($table, $order_by = '', $desc = false, $limit = false, $offset = false)
		{
			if ($this->tableExist($table))
			{
				$query = "SELECT " . $this->getColumnsForTable($table) . " FROM " . $table;

				if ($order_by)
				{
					if ($this->fieldExist($order_by, $table))
					{
						$query .= ' ORDER BY '. $order_by;
						if ($desc) 
						{
							$query .= ' DESC';
						}
					}
				}

				if ($limit !== false)
				{
					$query .= ' LIMIT :limit';
					if ($offset !== false)
					{
						$query .= ' OFFSET :offset';
					}
				}

				$req = $this->bdd->prepare($query);

				if ($limit !== false)
				{
					$req->bindParam(':limit', $limit, PDO::PARAM_INT);
					if ($offset !== false)
					{
						$req->bindParam(':offset', $offset, PDO::PARAM_INT);
					}
				}

				$req->setFetchMode(PDO::FETCH_ASSOC);
				$req->execute();
				return $req->fetchAll();
			}
		}
```

memento, 2015-10-12:
Le code du fichier controllers/internalConsole.php n'a pas été mis à jour avec le nouveau nom de méthode dans les versions ultérieures.
Je suppose que ça date de votre mise à jour du code pour qu'il soit plus en phase avec la dernière version de votre Framework Descartes.

memento, 2015-10-12:
Je présume également que vous vouliez remplacer la fonction getAll() par getFromTableWhere() ?

memento, 2015-10-12:
ça marche maintenant que je remplace getAll par getFromTableWhere dans controllers/internalConsole.php
Vous savez ce qu'il vous reste à faire ;)

Bonne journée !

RaspbianFrance, 2015-10-12:
Hop, c'est fixé !
En voilà un qui était passé entre les mailles du filet !

Merci du retour super efficace, comme d'habitude !

closed 2015-10-12 by RaspbianFrance