RaspbianFrance/raspisms #30: API Send: Envoie sms depuis PC avec URL

Nico80, 2015-12-29:
Bonjour,
J'ai installé Raspisms sur mon pi, j'arrive à programmer des SMS directement avec Raspisms:
http://192.168.1.35/RaspiSMS/scheduleds/

J'aimerai pouvoir envoyer/Programmer des SMS depuis mon PC (avec une macro Excel). Je n'arrive pas à envoyer de SMS avec une URL (j'ai essayé plusieurs possibilités, mais rien ne marche...):
http://192.168.1.35/RaspiSMS/smsAPI/send/?email=***********@yahoo.fr/&password=*******/&numbers=06*********/&text=test/
J'ai une erreur:
Not Found
The requested URL /Raspisms/smsAPI/send/ was not found on this server.
Apache/2.4.10 (Raspbian) Server at 192.168.1.35 Port 80

Ou existe-t-il une autre solution simple pour programmer un SMS depuis PC (Excel)...

Merci de votre aide, et bonnes fêtes...

memento, 2015-12-29:
Bonsoir, quelle version de RaspiSMS as-tu ?
Quand et comment (téléchargement du .deb, git clone) as-tu téléchargé le projet ?

Nico80, 2015-12-29:
Bonsoir, J'ai installé la version Raspbian Jessie du 21/11/15 la semaine dernière (Nouveau pi)
J'ai installé un serveur plex (Ça marche pas trop mal)
J'ai installé Raspisms v2 samedi dernier, avec le paquet deb comme indiqué sur raspbian-france.
Merci...

memento, 2015-12-29:
As-tu fais un tour dans les problèmes [résolus](https://github.com/RaspbianFrance/RaspiSMS/issues?q=is%3Aissue+is%3Aclosed) ?

Et plus précisément, en ce qui te concerne, regarde [ici](https://github.com/RaspbianFrance/RaspiSMS/issues/26) :

Il faut enlever le send/
La documentation n'est pas à jour. J'ai signalé le problème il y a quelques mois. J'espère que ton cas va motiver les troupes à modifier la doc en ligne. ;)

Nico80, 2015-12-29:
Je l'avais lu et testé malheureusement sans succès...
je viens de retester au cas ou.. mais marche toujours pas
http://192.168.1.35/Raspisms/smsAPI/?email=**********@yahoo.fr/&password=**********/&numbers=062***********/&text=test/
(C'est bien un "/?" après le smsAPI?)

J'ai aussi essayé de modifier les htaccess, sans succès dans le fichier /etc/apache2/apache2.conf
<Directory />
	Options FollowSymLinks
	AllowOverride All
	Require all denied
</Directory>

<Directory /usr/share>
	AllowOverride All
	Require all granted
</Directory>

<Directory /var/www>
	Options Indexes FollowSymLinks
	AllowOverride All
	Require all granted
</Directory>

Ci-joint mon fichier apach2.conf (en .txt pour l'upload)
[apache2.txt](https://github.com/RaspbianFrance/RaspiSMS/files/74292/apache2.txt)

Merci encore beaucoup pour l'aide..

memento, 2015-12-29:
Je teste ça demain, je n'ai pas accès à mon système ici.

Nico80, 2015-12-29:
Merci, je viens de télécharger l'image de raspisms pour la mettre sur une carte SD et comparer les fichiers apach2.
En remplaçant le fichier apach2.conf par celui d'origine de l'image.
RaspiSMS marche mais toujours  pas les URL et mon serveur plex ne marche plus...
Je débute dans linux donc j'ai peut-être zappé un truc simple...
Bonne soirée.

memento, 2015-12-30:
Salut,

Je fais quelques tests.
Mais je peux déjà te dire que ton url est incorrecte. Les majuscules comptent et il faut mettre ___RaspiSMS/___ au lieu de ___Raspisms/___ .

memento, 2015-12-30:
Attention aux ___/___ à la place des ___\___ .

memento, 2015-12-30:
Par exemple, dans ton cas, ça donnerait : 

http://192.168.1.35/RaspiSMS/smsAPI/?email=**********@yahoo.fr\&password=**********\&numbers=06********\&text=test

memento, 2015-12-30:
Au fait, c'est une requête GET.. enlève les antislashs (\) devant les &.

Ce qui donne chez toi : http://192.168.1.35/RaspiSMS/smsAPI/?email=**********@yahoo.fr&password=**********&numbers=06********&text=test

Tu devrais avoir un joli :
```
{"error":0}
```
 Qui signifie ___{"error":0}___

Pour rappel

```

Retour	Description
{"error":0}	Le SMS a été créé avec succès.
{"error":1}	Les identifiants fournis sont incorrects
{"error":2}	La création du SMS a échoué
{"error":3}	Des arguments obligatoires sont manquants
```

RaspbianFrance, 2015-12-30:
Merci beaucoup d'avoir pris le temps de répondre à Nico Memento (et pour ton implication en général d'ailleurs).
Nous avons réglé le problème dans la documentation !

closed 2015-12-30 by RaspbianFrance

Nico80, 2015-12-30:
Merci beaucoup memento, ça marche enfin!!!! c'est super sympa!!!
Bonnes fêtes de fin d'année.

memento, 2015-12-30:
De rien, bonne fêtes de fin d'années à tous !