RaspbianFrance/raspisms #33: Webhook pour recevoir les SMS

EthanSbbn, 2016-01-06:
Bonjour,

Je ne trouve pas de fonctionnalité pour appeler un point d'API externe à la réception d'un message SMS.
Ce webhook serai super utile pour intégrer entièrement RaspiSMS avec des serveurs et API externes.

Des plans pour cette idée ? Des hints pour le réaliser ?

memento, 2016-01-06:
> Je ne trouve pas de fonctionnalité pour appeler un point d'API externe à la réception d'un message SMS.

Bonjour, 

Peux-tu être plus précis ? On peut prendre ta requête dans plusieurs sens. Quel est ton scénario ?

RaspbianFrance, 2016-01-06:
Salut,
actuellement il n'existe pas de fonctionnalité de ce type.
Nous envisageons d'ajouter une API REST complète à RaspiSMS avec la version 3.0 (nous n'avons pas encore commencé à travaillé dessus).

Avec une telle API, un webhook ne serait plus vraiment nécessaire (il serait possible de récupérer régulièrement les derniers messages, ce qui reviendrai un peu au même).

Il faudra donc probablement attendre une version 3.0 !

closed 2016-01-06 by RaspbianFrance