RaspbianFrance/raspisms #35: Gestion plus poussée des contacts

romainguerrero, 2016-01-19:
Bonjour,

Pour des besoins personnels je souhaiterais avoir une gestion des contacts et des groupes un peu plus poussée :
  * Je vais avoir de besoin dans un premier temps de plus d'informations pour chaque contact telles que :
    * son prénom (dissocié du champ "name" actuel)
    * sa civilité (M./Mme)
    * sa date de naissance
    * sa situation amoureuse : célibataire/en couple
  * J'aurai également besoin de pouvoir gérer les groupes avec une hiérarchisation : pouvoir intégrer des groupes dans des groupes. Par exemple permettre de regrouper dans un groupe C toutes les personnes du groupe A et du groupe B.
  * Enfin je souhaiterais mettre en place des groupes "intelligents" grâce aux information rentrées dans les contacts afin de pouvoir créer des groupes qui seraient gérés de manière transparente tels que "Toutes les personnes entre 12 et 25 ans", ou encore "Tous les hommes célibataires" par exemple.

Est-ce que ce sont des choses qui sont prévues dans une nouvelle version de l'application ? Si ce n'est pas prévu, je me propose pour les mettre en place et partager le code sur votre projet si il vous convient !
Si cela peut vous intéresser, je suis à l'écoute de vos propositions de mise en place afin que cela puisse coller avec votre vision de l'application.

Merci par avance

RaspbianFrance, 2016-01-19:
Bonjour,
actuellement ces fonctionnalités ne sont pas prévues.

Si tu te sent le courage de développer ces différentes fonctionnalités, nous serions ravis de les intégrer.

L'idée qui guide notre travail sur RaspiSMS est toujours la simplicité d'utilisation.
Nous souhaitons que l'interface propose en priorité les fonctions essentielles et reste le plus simple possible. Cela se traduit notamment par la possibilité d'activer ou désactiver certaines fonctionnalités. Les fonctionnalités que tu cites devrait donc êtres activables, et désactivées par défaut.

Par ailleurs, nous souhaiterions que les modifications ici présentées ne viennent pas modifier les tables déjà existantes de la base de données. Il faudrait donc créer une nouvelle table pour les informations de l'utilisateur et pour les groupes hiérarchisés.

Pour ce qui concerne les groupes hiérarchisés, je penses que tu peux les inclure directement au sein du système actuel de groupes.

Pour ce qui concerne le style de programmation, il n'existe pas vraiment de lignes directrices officiels (pour l'instant, un jour peut-être aurons-nous le courage d'écrire tout ça). Il faudra que tu regarde le style actuel et que tu t'adaptes (après un petit tour sur ton GitHub, je pense que tu ne devrais pas avoir trop de mal)...

Il n'existe qu'une règle pas forcément évidente à voir à l’œil nu si on ne la connais pas déjà, c'est que nous essayons toujours de ne pas imbriquer les "if".
En gros, l'idée c'est que tout les cas qui ne correspondent pas au cas nominal peuvent presque toujours se traiter dans un "if" qui retourne un résultat pour sortir de la méthode. Chaque comportement "imprévu" est traité dans un "if" qui va soit récupérer le problème pour le faire revenir à une situation normal, soit mettre fin à la méthode.

romainguerrero, 2016-03-10:
Bonjour,

Comme vous avez du le voir dans la pull request #38 je vous ai soumise le mois dernier, j'ai pu mettre en place la gestion plus poussée des contacts. Je suis en train d'utiliser cette fonction chez moi avant de commencer les autres développements prévus sur les groupes afin de vérifier que tout répond correctement à mes besoins.
Justement à travers mon utilisation réelle, je me retrouve embêté sur quelques points :

1.  Le plugin magicSuggest limite par défaut les sélections à 10 résultats sauf à spécifier dans sa configuration `maxSelection: null` ce qui par extension empêche de créer des groupes de plus de 10 contacts. Est-ce volontaire de votre part de conserver cette limitation ou est-ce que vous souhaitez que j'intègre cette modification à ma pull request ?

1. Quelque chose de tout bête : les boutons d'ajout (de contact par exemple) sont positionnés en dessous du tableau qui liste toutes les entrées existantes. Problème : avec mes 500 contacts, il me faut scroller un petit moment avant d'arriver à ce bouton. Est-ce qu'il est prévu d'intégrer une pagination sur cette page de contacts comme cela peut être le cas sur d'autres (liste des SMS envoyés par exemple) ? A moins de placer le bouton (ou un doublon) au dessus du tableau pour palier à ce problème ?

1. Ensuite, avec 500 contacts, je me heurte aussi à un autre problème : celui de la recherche et de l'affichage de mes contacts qui n'est pas facilité par l'affichage trié par id. Est-ce qu'il est prévu une fonction de recherche/tri sur ce tableau dans vos développements à venir ? Si ce n'est pas le cas, j'ai trouvé un plugin JS (http://www.listjs.com/) qui permet de trier, filtrer et rechercher dans un tableau que je pourrais mettre en place si vous le souhaitez. Ce plugin gère également la pagination ce qui permettrait au passage de résoudre mon problème ci-dessus !

1. Enfin, je me suis aussi rendu compte que pour envoyer un SMS à tous les contacts enregistrés, ils doivent être soit ajoutés un à un au SMS soit faire tous partie d'un ou plusieurs groupes (ce qui implique là aussi de les avoir ajouté un à un à ce(s) groupe(s) ce qui avec 500 contacts n'est pas gérable...) et d'ajouter ces derniers. Je pense rajouter une case à cocher dans la création d'un SMS : "Envoyer à tous les contacts" qui désactiverait la saisie des champs existants (numéro, contact, groupe). Qu'en pensez-vous ?

RaspbianFrance, 2016-03-10:
Salut,
merci de ces super retours.
Nous n'avons pas encore eu le temps d'intégrer ta pull request (nous le
ferons sans doute début avril, je ne pense pas avoir le temps avant).

Concernant les points soulevés :

   1. Pour magicSuggest, nous ne nous étions jamais rendu compte de cette
   limite, il faudrait effectivement corriger cela !
   2. Pour la pagination il serait effectivement pas mal d'en rajouter une,
   basée sur le code de celle des contacts par exemple.
   3. Pour le tri, il faudrait également ajouter quelque chose, mais la
   pagination en JS ça ne me tente pas trop.
   4. L'ajout d'un champ d'envoi à tous les contacts est une bonne idée.

Merci encore de ton investissement sur ce projet, c'est dernier temps je
crois que tu codes plus que nous ;)

romainguerrero, 2016-03-18:
Salut,

Je viens de pousser la modification pour magicSuggest ainsi que la correction d'un bug que j'ai rencontré lorsque j'ai essayé de modifier un SMS qui ne comportait pas de numéro ni de contact.
A ce sujet, est-ce volontaire d'afficher à la création d'un SMS un champs de numéro vide ? En effet, si je ne m'en sers pas et que je ne le remplis pas non plus, je me retrouve avec un message d'erreur (non bloquant) qui me dit que certains numéros ne sont pas valides.

Enfin, dites-moi si vous préférez aussi qu'on continue nos discussions ici ou si vous préférez qu'on fasse ça par mail.

A bientôt !

memento, 2016-04-11:
Salut, je n'ai pas trop le temps de coder en ce moment sur ce projet mais vos conversations sont très intéressantes et qui sait quelqu'un d'autre pourrait vous aider sur un point ou vous suggérer quelque chose. Ce n'est pas une démocratie mais si je devais voter, je voterais pour que vous poursuiviez vos conversations ici !

Merci !

Bien à vous,

romainguerrero, 2016-04-11:
Salut !

Pas de soucis pour moi de continuer nos échanges ici ;)

Je viens de pousser l'ajout de la fonction d'envoi de SMS à tous les contacts du répertoire ainsi qu'une petite modification pour permettre d'accepter les SMS "Stop" qui ne sont pas en majuscules (je sais pas si c'est moi mais mes contacts ne l'envoient jamais en majuscule...).

Pour revenir à la question de la recherche/filtre/tri, il ne sera à mon avis pas possible de le mettre en place côté JS si la pagination se fait côté PHP (qui sera autrement plus complexe surtout sans l'utilisation d'un framework répandu). Je suis pleinement conscient qu'étant à l'initiative du projet, il est logique que vous ayez la main sur les choix technologiques à implémenter ou non et c'est pour cela que je souhaiterais connaitre les raisons qui font que la pagination JS ne vous tente pas trop. Est-ce que vous craignez un problème de lenteur à l'exécution ? Ou bien un problème de compatibilité navigateur ?

A bientôt !

romainguerrero, 2016-04-11:
Et autre petit ajout : les transferts de SMS par mail incluent maintenant dans le titre le nom du contact associé si jamais il est trouvé en base.

tux-vador, 2016-05-26:
Bonjour,

Excellent projet et discussion passionnante :

Romain est ce que je peux avoir ta version avec tes amélioration qui m'intéresse fortement ?

Si ils sont deja intègre ici comment je peux les mettre sur ma Raspberry.

Merci pour votre travail et réponse d'avance

romainguerrero, 2016-06-01:
Bonjour @tux-vador,

J'ai décrit la procédure dans cette discussion : [https://github.com/RaspbianFrance/RaspiSMS/commit/a6f849cfcefe87bf57d6376fac4addcea614a502#commitcomment-17695034]
Je te laisse voir si ces informations te suffisent sinon n'hésites pas à me demander !

@+
Romain

tux-vador, 2016-06-01:
Merci pour ton aide @romainguerrero 

Je teste ça et je reviens vers toi

closed 2021-02-14 by OsaAjani