RaspbianFrance/raspisms #36: Connexion impossible ( 404 Not Found )

Loup-des-Neiges, 2016-01-22:
Hello,

Installation fraîche depuis la version "image", connexion à l'interface web
impossible. J'arrive bien sur la page de connexion, mais après validation
des identifiants, je me retrouve sur une page : "/RaspiSMS/connect/login/"
qui me retourne une erreur 404 Not Found.

Raspberry 1 b+

RaspbianFrance, 2016-01-23:
Salut,
on a fait une petite mise à jour il y a un ou deux jours et nous avons
peut-être fait une mauvaise manipulation, nous vérifions ça tout de suite.

Cordialement,
Raspbian France

RaspbianFrance, 2016-01-23:
Ok, c'est les mots de passe que nous avons changer sans faire attention.
Il faut utiliser "admin@example.fr" et "admin".

Par ailleurs, en cas de mot de passe incorrect la redirection n'est pas
bonne. Néanmoins, ça ne sera pas fixer avant quelques jours je penses.

Sympol, 2016-03-01:
Pour la redirection, voici comment j'ai reglé chez moi
`public function connection()
		{
			
			//Creation de l'object de base de données
			global $db;
			
			$email = $_POST['mail'];
			$password = $_POST['password'];

			if (!$users = $db->getFromTableWhere('users', ['email' => $email]))
			{
				$_SESSION['errormessage'] = 'Vous n\'existez dans notre base de donnée .';
				header('Location: ' . $this->generateUrl('connect'));
				return false;
			}

			$user = $users[0];

			if (sha1($password) != $user['password'])
			{
				$_SESSION['errormessage'] = 'Vérifiez vos paramètres de connexion.';
				header('Location: ' . $this->generateUrl('connect'));
				return false;	
			}

			$_SESSION['connect'] = true;
			$_SESSION['admin'] = $user['admin'];
			$_SESSION['email'] = $user['email'];
			$_SESSION['transfer'] = $user['transfer'];
			$_SESSION['csrf'] = str_shuffle(uniqid().uniqid());
			header('Location: ' . $this->generateUrl(''));
			return true;
		}`

RaspbianFrance, 2016-03-07:
Salut,
c'est déjà fix sur le github mais ça ne sortira probablement pas avant la prochaine version de RaspiSMS.

closed 2016-03-07 by RaspbianFrance