RaspbianFrance/raspisms #37: API issue

Squal78, 2016-01-30:
Bonjour,
Pour commencer merci pour ce support boulot !

Ensuite, j'essaie d'utiliser les API, mais j'ai quelques problemes avec les numéros/groupes/contacts

une syntaxe comme ci-dessous me renvoie une erreur : curl: (3) [globbing] bad range specification in column 107

curl   http://192.168.0.9/RaspiSMS/smsAPI/?email=email@email.fr\&password=test\&date=2016-01-30%2022:28\&contacts[]=pro\&contacts[]=perso\&text=Texte%20du%20SMS%20à%20envoyer;

Idem, lorsque j'utilise les groups.
curl   http://192.168.0.9/RaspiSMS/smsAPI/?email=email@email.fr\&password=test\&date=2016-01-30%2022:28\&groups[]=groupedetest35\&text=Texte%20du%20SMS%20à%20envoyer;

Par ailleurs, lorsque j'essaie de mettre plusieurs numéro à envoyer, seul le premier reçoit le sms.

Merci de votre aide,

RaspbianFrance, 2016-03-07:
Salut,
il faut que tu utilises la commande "curl -g" au lieu de "curl".

closed 2016-03-07 by RaspbianFrance

Nems27, 2017-12-14:
Salut,

Désolé de revenir dessus mais je rencontre un soucis
:frowning: … Lorsque je lance cette commande dans la console < curl -g http://sms/RaspiSMS/smsAPI/?email=XXX@XXX.fr\&password=XXXXX\&numbers=06XXXXXXXX\&contacts=toto\&groups[]=groupe\&groups[]=test\&text=TestMultiGroupe > Seul “numbers” reçoit le message mais le contact et les groupes non… Avez-vous des idées?

Merci d’avance de votre aide.