RaspbianFrance/raspisms #43: limitation nb contact par groupe

Blacksly, 2016-04-23:
Bonjour, 

Déjà merci beaucoup pour raspi SMS, c'est au top ! 
Cependant, je me demandais si l'on pouvais enlever la limite de 10 contact par groupe ? merci bien :)

romainguerrero, 2016-04-23:
Bonjour,
J'ai levé la limitation de mon côté et la modification est inclue dans la pull request que j'ai faite et qui est en attente de validation ;)

Blacksly, 2016-04-23:
Ha génial ! je te remercie :) les sms ne galère pas trop a s'envoyer sur des gros groupe ? ^^

romainguerrero, 2016-04-23:
Je n'ai pas rencontré de problème particulier, même si je t'avoue que je n'ai pas forcément analysé tous les status de SMS (un de mes groupes compte 500 contacts...) ;)
Ca pourrait d'ailleurs être une idée d'ajout : un rapport sur le nombre de SMS programmés vs délivrés !

Blacksly, 2016-04-23:
c'est vrai que ça pourrais être sympa ^^ je ne suis pas très familier a github, mais du coup cela peut prendre combien de temps avant tes modifs soit validé ? peut-on les récuperer des maintenant ? thanks ;)

romainguerrero, 2016-04-23:
Sur GitHub (comme les autres services du genre) pour pouvoir contribuer à un projet il suffit de le "copier" dans son compte et faire les modifications dessus. Ensuite on demande aux personnes qui gère le projet si elles veulent bien accueillir dans leur code officiel, les modification qu'on y a apporté. Pour ce qui est du temps que cela prends, c'est à Raspian France qu'il faut poser la question ;)
Il leur faut le temps d'analyser le code pour vérifier que rien ne puisse poser problème et que les modifications leurs conviennent. Ensuite ils peuvent accepter ou demander des ajustements avant de le faire. Quand c'est validé, tout le monde aura ensuite accès à ces nouveautés ou corrections. Voilà comment fonctionne GitHub ;)

Blacksly, 2016-04-23:
Okay ça marche ;) merci encore

closed 2016-04-23 by Blacksly