RaspbianFrance/raspisms #44: Contact romain GUERRERO

Blacksly, 2016-04-24:
Bonjour, 
Désolé de te contacter comme cela, mais crois tu que l'on pourrais se contacté en privé ? 

Merci bien :)

memento, 2016-04-25:
Bonjour, 
Oui, c'est à quel sujet ?

:)

Blacksly, 2016-04-25:
Bonjour,

Merci de m'avoir répondu et merci encore des modifications que vous avez
apportées a raspiSMS et que j'ai réussi à prendre ^^

En essayant de vous contacter à ce sujet hier (d'ailleurs, je n'ai pas
trouvé comment envoyer des messages privé sur github ^^ ), j'ai remarqué
que vous étiez chef de projet dans une entreprise à Aix en Provence. En
fait je recherche actuellement une entreprise pour faire de l'alternance et
je me demandais si je pouvais vous envoyer un CV ou si vous connaissiez des
entreprises qui recrutaient ?

Merci beaucoup,
Nicolas LOBELLO

closed 2016-05-05 by Blacksly

romainguerrero, 2016-05-06:
Bonjour Blacksly,

Désolé pour le temps de réponse. Oui je suis chef de projet technique dans une agence de développement web et mobile à côté d'Aix-En-Provence : Acseo Conseil. Tu peux aller faire un tour sur notre site Internet : http://www.acseo-conseil.fr ou nous envoyer ta canditature à contact@acseo-conseil.fr.

A bientôt !