RaspbianFrance/raspisms #45: SMS envoyé plusieurs fois

max06200, 2016-04-29:
Bonjour,

Lorsque j'envoie un SMS via RaspiSMS, celui-ci est envoyé 2, voir 3 fois de suite....

Auriez vous une idée?

Merci

RaspbianFrance, 2016-05-02:
Salut,
vérifie la crontab de l'utilisateur root. Tu trouveras probablement des
lignes dupliquées.

dam09fr, 2016-05-02:
Dans l'éventualité d'une double exécution du cron, ne serait-il pas possible de mettre en place un contrôle de la façon suivante, afin de sécuriser le fonctionnement : 

Dans le script du cron :
si le fichier "cron.pid" existe alors
-  si ce fichier contient un PID en cours d'exécution alors
-  -  exit (la tâche est en cours)
-  sinon
-  -  on supprime le fichier "cron.pid" (le processus s'est fermé anormalement)
-  finsi
finsi
on crée un fichier "cron.pid" dans lequel on place le PID du processus courant
(exécution normale du script ...)
on supprime le fichier "cron.pid" à la fin de l'exécution

RaspbianFrance, 2016-05-02:
Ce serait effectivement une possibilité.
De toute façon, le mécanisme de cron risque d'être pas mal revu
prochainement sur RaspiSMS.
Nous aimerions bien faire un système plus proche d'un deamon ou quelque
chose comme ça.

StephaneTy-Pro, 2016-09-07:
Bonjour,
J'ai le même problème mais il est complètement aléatoire (au début j'ai pensé qu'il s'agissait d'un problème :" d'ordre de saisie de champs" mais au final il semblerait que non)

Auriez vous des idées pour corriger le sujet en l'état (=>avec l'utilisation de cron)

D'avance merci

memento, 2016-09-07:
Bonjour,
Pour vérifier la table cron, tu peux taper la commande suivante (en root) :

'''
crontab -e
'''

Et vérifier les lignes en double.
Tu dois normalement avoir quelque chose comme ça :

'''
* * * * * cd /var/www/RaspiSMS && /var/www/RaspiSMS/console.php -c parseReceivedSMS
* * * * * cd /var/www/RaspiSMS && /var/www/RaspiSMS/console.php -c sendScheduleds
'''

Une fois la vérification/édition terminée, comme sous l'éditeur nano, fais un Ctrl+x
Si tu as fait une modification, on te proposera te taper o pour sauvegarder ou n pour abandonner les modifications.

StephaneTy-Pro, 2016-09-13:
Merci je vais regarder au prochain envoi

Tindome, 2016-09-29:
@memento Tout s'explique j'avais les lignes que tu évoques mais 4fois j'ai tout viré pour n'en laisser que deux
Merci

max06200, 2016-10-10:
Je reviens sur le sujet, car j'avais laissé tombé un moment, mais je retente. Alors je n'avais rien dans mon fichier Cron, j'ai donc mis les 2 ligne indiqué plus haut, mais cela ne change rien.

Lorsque j'envoie un sms depuis le dashboard, certaine fois il l'envoi bien une fois, et certaine fois 2, 3 voir 10 fois.

Une autre idée pour solutionner le problème?

RaspbianFrance, 2016-10-31:
Salut,
étais-tu bien en mode root quand tu à lancer la commande crontab -e ?
Lance d’abord la commande "sudo -s", puis seulement après "crontab -e".

Cordialement,
Raspbian France

memento, 2016-11-30:
@max06200 Salut, as-tu solutionné ton problème ?

vincent1890, 2019-09-21:
J'ai le même problème dans **http://IP/RaspiSMS/sendeds/** je me retrouve avec deux ligne pour chaque message et donc une des deux en toujours en **statut Inconnu** et l'autre en délivré ce qui est logique car je reçois un seul SMS.

Et il y bien que 2 ligne dans le crontab
Si quelqu'un a trouver la solution je suis preneur.

closed 2021-02-14 by OsaAjani