RaspbianFrance/raspisms #46: Signal strength

sebecam2000, 2016-05-20:
Il n'y a actuellement pas moyen de voir la force du signal, ce qui est embêtant lorsque certains sms ne sont pas envoyés à cause d'un signal trop faible.
Pour pallier ce problème, j'utilise la commande comgt qui me renvoit le résultat d'un AT+CSQ (à mettre en lien avec le tableau de http://m2msupport.net/m2msupport/atcsq-signal-quality/ ), et je vérifie le nombre de messages en attente dans l'outbox.
Est-il prévu d'intégrer un tel composant dans votre interface?

tux-vador, 2016-06-01:
Je pense que cette demande est vraiment utile aussi

memento, 2016-06-01:
Pareil. Je suis pour.

RaspbianFrance, 2016-06-07:
Salut,
nous notons l'idée et nous essayerons d'ajouter cette fonctionnalité dans
la prochaine version !

closed 2021-02-14 by OsaAjani