RaspbianFrance/raspisms #47: Error at init connection : DEVICENOTEXIST[4] - Gammu (mais posant problème aux utilisateurs de RaspiSMS)

memento, 2016-06-01:
Bonjour à toutes et tous,

J'ai un problème avec gammu-smsd mais utilisant le matériel conseillé sur le site de RaspiSMS (clé 3G HUAWEI E220), je poste ici et je donnerai la solution pour les utilisateurs de RaspiSMS.
De temps à autres, la communication entre gammu et la clé 3G est perdue.
Je poste ici aussi parce que je pense que la solution est de créer un lien persistant vers la clé 3G (dans l'installation du .deb de RaspiSMS ?).

`May 31 06:25:38 mamachine gammu-smsd[1833]: Starting phone communication...
May 31 06:25:38 mamachine gammu-smsd[1833]: Error at init connection: Erreur à l'ouverture du périphérique: il n'existe pas. (DEVICENOTEXIST[4])
May 31 06:25:38 mamachine gammu-smsd[1833]: Going to 30 seconds sleep because of too much connection errors
May 31 06:26:01 mamachine /USR/SBIN/CRON[28366]: (root) CMD (cd /var/www/RaspiSMS && /var/www/RaspiSMS/console.php -c sendScheduleds)
May 31 06:26:01 mamachine /USR/SBIN/CRON[28368]: (root) CMD (cd /var/www/RaspiSMS && /var/www/RaspiSMS/console.php -c parseReceivedSMS)
`

memento, 2016-06-01:
Quelqu'un est déjà passé au lien symbolique vers sa clef Huawei (ou autre) pour utiliser RaspiSMS ?

octera, 2017-01-17:
Je suis tombé sur ce post par hasard, deux chose que je peux partager : 

-  Les clef huaway ne sont pas stable du tout... elle plante fréquement, j'utilise un crontab avec ce script pour les reboot toutes les 30 minutes : https://gist.github.com/octera/7f5a4acd944b880361d3e2ebe8fcda25 (trouvé sur stackOverflow)
==> A lancer avec ./usbreset.sh Huawei

- Pour eviter tout changement de port tty, j'utilise l'accès par id (qui est un symlink vers les ttyUSBxx) : /dev/serial/by-id/usb-Huawei_Technologies_HUAWEI_Mobile-if00-port0

extrablind, 2017-06-21:
Pas eu de ptoblème de ce genre de mon côté. 
Par contre comme le mentionne @octera, ne faites pas référence à vos dongle 3g via le traditionnel /dev/USBX, au démarrage les tty peuvent s'inverser; faites référence au port physique !

closed 2021-02-14 by OsaAjani