RaspbianFrance/raspisms #48: Envoi SMS à partir d'un site web

yaroing, 2016-06-02:
Bonjour,

Je suis aux anges depuis quelques jours, après la découverte de cet excellent produit. Simple, rapide et ergonomique. Différent de tous ce que j'avais imaginé.

Pour tout vous dire, nous sommes dans une phase d'adoption active pour le suivi de nos actions humanitaires au Niger.

Ma question : 
Comment envoyer un SMS à partir d'un site web (via php) :
- en utilisant l'API;
- en insérant le message dans les tables **scheduleds** et **scheduleds_numbers**.

J'avoue que pour la dernière option, je suis bloqué, question de synchronisation.

Merci pour votre éclairage.

RaspbianFrance, 2016-06-07:
Salut,
nous sommes ravis de voir RaspiSMS utilisé dans ce cadre ! Pourrais-tu nous
dire de quelle association il s'agit ?

Pour l'envoi de SMS via PHP, le plus simple est effectivement d'utiliser
l'API. L'autre solution serait de ne pas utiliser l'API, mais de simuler
l'utilisation du formulaire normal en PHP.
Dans la version 3 nous rajouterons une API REST complète qui devrait
faciliter ce type d'intégrations !

Cordialement,
Raspbian France

yaroing, 2016-06-10:
Bonjour, 

Je travaille pour une agence humanitaire (unicef).

Cet outil nous sert de point de contact avec les bénéficiaires de certains de nos interventions.

Merci pour cette précision. Nous attendons vivement la version 3.

Cordialement.

closed 2016-06-10 by yaroing