RaspbianFrance/raspisms #49: Gestion multiple modems

pierrediancourt, 2016-06-08:
Bonjour,

Tout d'abord merci pour ce super projet !

J'ai une proposition de feature à vous soumettre : la gestion de plusieurs téléphones/modems connectés simultanément sur le raspberryPi

Cela me semble aisément réalisable en raison de ces lectures : 
[doc smsd a propos du multi-modem](https://wammu.eu/docs/manual/smsd/config.html#smsd-multi) où l'on voit qu'il suffit de passer le paramètre -c pour spécifier le fichier de config correspondant au device désiré
[doc smsd-inject](https://wammu.eu/docs/manual/smsd/inject.html) où l'on voit que ce même paramètre existe aussi pour la commande que vous utilisez actuellement dans votre code.

Ce qui me fais me dire que dans : controllers/internalConsole.php

```
//Commande qui envoie le SMS
$commande_send_sms = 'gammu-smsd-inject TEXT ' . escapeshellarg($number) . ' -report -len ' .mb_strlen($text_sms) . ' -text ' . $text_sms;
```

Il suffirait de rajouter l'option "-c filename" pour spécifier le fichier de configuration concernant le device branché que l'on voudrait utiliser pour envoyer le sms.

De nombreuses possibilités découlent de ça, comme le fait de pouvoir spécifier quel device on veut utiliser pour "smser" tel ou tel contact, ou cela nous est indifférent. On peut aller jusqu'à imaginer une gestion fine du nombre de messages envoyés par device (donc ligne/abonnement), la fréquence maximale d'envoi par heure etc...  

Tout cela permettant d'avoir un contrôle plus fin sur les numéros et abonnements utilisés pour l'envoi de sms. Je pense que cela pourrait rendre service à quiconque compte utiliser ce projet pour de grands volumes de sms ou même pour de l'envoi de faible volume mais dans de multiples destinations à l'international (optimisation des frais suivant le device et donc abonnement utilisé, basé sur le préfixe du pays). 

Je pense que c'est une des rares fonctionnalités encore manquantes à ce projet =)
Merci encore pour celui-ci !

memento, 2016-06-13:
Salut,

J'ai déjà tester mon installation avec deux modems Huawei E220. La première difficulté rencontrée quand on utilise un Raspberry pi (1, 2), c'est l'alimentation des périphériques USB (ampérage). En effet, l'utilisation de chaque périphérique va pomper un certain courant et littéralement déconnecter d'autres périphérique reliés à ta carte Raspberry Pi. 

Par exemple, le modem Huawei E220 consomme 2.5 Watts (doc)[http://www.sms-wiki.org/attachments/117/huawei-e220.pdf]. 

Attention donc à fournir ce qu'il faut niveau ampérage. Si tu as un Raspberry Pi 1 ou 2, tu devras doubler l'arrivée de courant vers les ports USB. C'est possible en modifiant le fichier /boot/config.txt et en y ajoutant la ligne :

```
max_usb_current=1
```

Remarque : Il n'y a aucun espace sur la ligne !

Ensuite, tu feras face à un petit soucis d'identification des périphériques. En effet, à chaque redémarrage de raspbian, tu n'as aucun ganratie de retrouver tes périphériques derrières la même "adresse physique" 
```
/dev/ttyUSB0
/dev/ttyUSB1
...
```

Or, je crois savoir que tu voudrais donner une rôle précis à chaque carte SIM (nationale/internationale, etc). J'ai eu ce problème et j'ai une solution en cours de tests. ça t'intéressera (peut-être)[https://github.com/RaspbianFrance/RaspiSMS/issues/47]

Bien à toi !
Memento

pierrediancourt, 2016-06-13:
Bonjour,

J'ai reçu mon model Huawei E169 ce matin et tout se passe dans le meilleur des mondes.
Merci pour cette réponse au contenu pertinent et instructif ! Je ne risquais pas de penser aux limitations électriques. Je possède un Raspberry Pi version 3 et ne pense donc pas être concerné mais peut-être que ces remarques m'éviteront de m'arracher les cheveux lorsque je recevrai mon deuxième modem !

Bonne journée !

NB : [Sur cette page](http://raspbian-france.fr/envoyer-sms-raspberry-pi/) il y a une correction à effectuer : remplacer la ligne 

>  RunOnReceive = /var/www/RaspiSMS/parseSMS.sh

par celle ci
> RunOnReceive = /var/www/html/RaspiSMS/parseSMS.sh

darKy123, 2016-06-30:
Sorry i'm new here and i not speak france :/ , but my question is i have edited the php ( controllers/internalConsole.php) to
$commande_send_sms = 'gammu-smsd-inject TEXT ' . escapeshellarg($number) . '  -len ' .mb_strlen($text_sms) . ' -text ' . $text_sms;
without - report and in the gammu config i have deliveryreport= no but in the webinterface discuss with number 4912345678 .. it comes a icon after sending ??

Is that the icon that sms is sended or is this the "delivery report from my provider"?
Sorry for that but i will send in my holidays sms but when delivery reports my money goes over (300 sms O_o)

big thanks and sorry i'm new here :)

closed 2021-02-14 by OsaAjani