RaspbianFrance/raspisms #50: Problème avec les accents lors d'un envoi php

Mafraysse, 2016-06-13:
Bonjour à tous.
J'ai trouvé un script permettant d'envoyer des sms en passant pas php. 
Le script est lancé sur un serveur A qui envoie la commande via curl au raspberry
Problème les accents ne sont pas transmis !!! Idem pour les apostrophes.

Voici le script qui crée le texto : 
`<!doctype html>
<html>
<head>
<meta charset="UTF-8">
<title>Envoi texto</title>
</head>

<body>
<?
function post_to_url($url, $data) {
   $fields = '';
   foreach($data as $key => $value) { 
      $fields .= $key . '=' . $value . '&'; 
   }
   rtrim($fields, '&');

   $post = curl_init();

   curl_setopt($post, CURLOPT_URL, $url);
   curl_setopt($post, CURLOPT_POST, count($data));
   curl_setopt($post, CURLOPT_POSTFIELDS, $fields);
   curl_setopt($post, CURLOPT_RETURNTRANSFER, 1);

   $result = curl_exec($post);

   curl_close($post);
   return true;
}

$data = array(
   "numero" => "06XXXXXXXX",
   "texte" => "L'été sera chaud."
);
//print_r ($data);
if(post_to_url("http://monserveur/sendsms.php", $data))
{
	echo "SMS envoyé";
}
?>
</body>
</html>`

Et voici le script qui gère l'envoi, sur le raspberry: 
`<!doctype html>
<html>
<head>
<meta charset="UTF-8">
<title>Document sans titre</title>
</head>

<body>
<?php
$numero = $_POST['numero'];
$texte = $_POST['texte'];
echo `sudo gammu-smsd-inject TEXT '$numero' -text '$texte'`;
?>
</body>
</html>`

2 jours que je cherche. J'offre une glace à celui qui trouve un piste.
Merci d'avance.

memento, 2016-06-13:
Salut !
Je ne vois pas trop le rapport avec le projet RaspiSMS.
Je pense que tu devrais plutôt poster à ta question sur  un site comme stackoverflow ou openclassroom.
Il doit s'agir d'un problème de filtrage des caractères spéciaux (') et pour les accents, probablement un problème d'encodage des caractères. Tu peux également regarder en ligne le manuel de la fonction php curl_setopt() et voir si tu dois appliquer des arguments particuliers à ta fonction ou des filtres, encore une fois. Tout est en ligne.

Sinon, si tu ne veux pas avoir à réinventer la roue, je te recommande d'utiliser RaspiSMS !
Tu peux même l'utiliser à partir de ton site perso grace à l'[API](http://raspisms.raspbian-france.fr/documentation/#api) et revenir nous voir si tu as des questions.

Excellente soirée !
Memento

memento, 2016-06-13:
Ravis d'avoir pu t'aiguiller !
Bonne soirée

closed 2021-02-14 by OsaAjani