RaspbianFrance/raspisms #56: Probleme Envoi Mail veille du dongle 3G ?

Marc78330, 2016-09-08:
Bonjour, 

Nous testons depuis une petite semaine maintenant la solution RaspiSMS. 

Elle fonctionne mais de temps en temps les sms ne partent plus pour une raison inconnue. 

Nous utilisons un dongle huawei E-372 et de temps en temps après une longue période d'inactivité les sms ne partent plus il faut lancer soit un gammu --identify ou relancer le raspberry complétement. 

Nous sommes un peu déroutés et ne savons pas ou chercher ?  avez-vous une idée ? Est ce que le dongle ce met en veille ? . Help

darKy123, 2016-09-10:
Answer in English :)
I have tested many modems Huawei 372 or Huawei E220 with activ usb hub.
All modems disconnect after few hours working time .. i have used Huawei e220 WITHOUT passiv hub, directly to the raspberry and it rocks since 80 days without problems.

extrablind, 2017-06-21:
Same here. No problem with modem Huawei E169 E169G : can run for hours without losing connection.
Check rights of your devs. Check signal strength too.

closed 2021-02-14 by OsaAjani