RaspbianFrance/raspisms #63: probleme reception sms

Blacksly, 2016-11-25:
Bonjour, 
Depuis 1 ou 2 jours il met impossible de recevoir des sms sur le raspi alors que ça a marcher pendant près de 6 mois ... De plus, l'envoie fonctionne parfaitement ... Une idée ? 

je ne trouva pas les log non plus
Merci

memento, 2016-11-25:
Bonjour,
Tu a peut-être l'équivalent d'un "bourrage papier" dans ta file de réception.
ça arrive quand un fichier vide traîne dans ta cette dernière.
Le code de RaspiSMS ne prévoit pas de gérer les fichier SMS vide (je parle du fichier pas du SMS en lui-même) car justement les fichiers vide pas sensé arriver.

Je te recommande de te connecter en SSH à ton Raspberry Pi, en temps qu'utilisateur root :
```
sudo su
```
Puis recherche tous les fichiers vide (de taille nulle) dans le répertoire /var/www/RaspiSMS/receiveds :
```
find /var/www/RaspiSMS/receiveds -size 0
```

Si cette commande ne te renvoie rien, c'est que le problème vient d'ailleurs.
Si par contre cette commande te renvoie un ou plusieurs fichiers alors il faudra le(s) supprimer.
Dans ce cas, tu peux le faire à la main ou alors exécuter cette commande :
```
find /var/www/RaspiSMS/receiveds -size 0 -print0 | xargs -0 rm
```

Aller plus loin dans le contournement :
Si tel était bien ton problème, en attendant que l'équipe patch le code pour qu'il supprime directement fichier vide au lieu de planter tout le processus de réception (la raison pour laquelle tu ne reçois plus rien), tu peux programmer une tâche cron qui fera le ménage par exemple une fois toutes les 10 minutes.

Si cette manip' t'intéresse, voici le détail :

Connecte-toi en root :
```
sudo su
```
Affiche la table cron de l'utilisateur root (un genre de planning des tâche, comme sur ton frigo) :
```
crontab -e
```
Ajoute la ligne suivante :
```
*/10 * * * * find /var/www/RaspiSMS/receiveds -size 0 -print0 | xargs -0 rm
```
Enregistre et quitte la table cron (Ctrl+X , O, Entrée)

Tu peux faire le choix d'un solution plus propre et lancer la commande à partir d'un script du genre :
```
*/10 * * * * /chemin/vers/ton/script/bourrage-papier.sh
```
Contenu du script bourrage-papier.sh : 
```
#!/bin/bash
find /var/www/RaspiSMS/receiveds -size 0 -print0 | xargs -0 rm
exit 0
```

J'espère avoir été assez clair.
Bonne soirée !

Blacksly, 2016-11-26:
Hello,
merci pour ta réponse, ce pdt la commande que tu ma donné me renvoie ceci : 
find /var/www/RaspiSMS/receiveds -size 0

Et te fait pas, je suis a l'aise sur linux ^^

memento, 2016-11-27:
Ok, pas de soucis. Je préfère toujours détailler au début. Qui peut le plus, peu le moins. Comme ça, je ne prends pas le risque de dégoûter les nouveaux venus sur linux (Raspbian est souvent leur première distrib linux).

> merci pour ta réponse, ce pdt la commande que tu ma donné me renvoie ceci :
find /var/www/RaspiSMS/receiveds -size 0

"find /var/www/RaspiSMS/receiveds -size 0", c'est la commande, mais qu'est-ce qu'elle renvoie ?
Rien ou un ou plusieurs chemins fichiers ?

Peux-tu copier/coller la trace console ?

Blacksly, 2016-11-30:
Hello, désolé de pas avoir répondu plus tôt, et surtout désole de t'avoir déranger pour rien car il se trouve que c'est la carte sim qui est morte ... je n'y avait pas pensez car cela marchait dans un sens, très étrange ...
en tout cas merci pour ton temps !

closed 2016-11-30 by Blacksly