RaspbianFrance/raspisms #66: contenu des sms reçu  = pending

Blacksly, 2016-12-12:
Salut a tous! 

Petite question, se recois certain sms avec pour contenue pending, a quoi est-ce du ? 
Merci

RaspbianFrance, 2016-12-12:
Salut,
il faudrait regarder du coté de la doc de gammu.

Cordialement,
Raspbian France

memento, 2016-12-12:
Bonsoir, J'ai déjà eu ce cas.
Les SMS reçus avec le texte "pending" concernent certains opérateurs comme orange et ses sous-traitants comme Coriolis telecom.
Cela veut tout simplement dire que le SMS est en attente de livraison.
Cela ne sert à rien.
On pourrait imaginer un système de filtrage au moment de lire les SMS reçus.
On peut aussi imaginer un encart en plus dans l'interface d'administration pour ajouter des masques de filtrage.

Blacksly, 2017-01-25:
hello ! punaise, j'ai pas vu ton commentaire depuis tout ce temps ... en tout cas merci pour l'info. 
Personnellement j'ai fais un truc très sale mais qui fonctionne, un cron qui supprime les sms "pending" ... en attendant de me remettre plus dans le code ^^

closed 2021-02-14 by OsaAjani