RaspbianFrance/raspisms #69: Utiliser le code source de RaspiSMS

Tichinfopro, 2017-01-04:
Bonjour,
j'ai lu la documentation de RasiSMS et je trouve votre travail très intéressant. Etant nouvelle dans le domaine de la programmation je cherche à reconstituer l'application en local puis-je avoir des pistes. 
Merci

RaspbianFrance, 2017-01-04:
Bonjour,
Tu peux trouver toutes les sources sur le github du projet ->
https://github.com/RaspbianFrance/RaspiSMS

extrablind, 2017-06-21:
@Tichinfopro 
RaspiSMS est basé sur du code MVC maison qui constitue un petit framework. Il faut donc que tu apprenne d'abord ce qu'est le MVC (Model View Controller) tu pourras ainsi t'y retrouver mieux dans le code. Va sur Open Classroom pour apprendre, par exemple.

Tichinfopro, 2017-09-06:
Merci. je l'ai fais et j'ai compris. Merci bien

closed 2021-02-14 by OsaAjani