RaspbianFrance/raspisms #71: Réception SMS impossible.

justin83, 2017-01-23:
Bonjour, 

Comme précisé dans le titre, je n'arrive pas à recevoir de sms sur RaspiSMS. Après avoir fait plusieurs tests afin d'identifier le problème, je ne trouve pas la solution à mon niveau. 

L'envoi depuis l'interface Web fonctionne très bien. 
Lors de la réception d'un sms, le message arrive bien sur le téléphone connecté (Samsung Solid 2710) puis arrive bien dans le dossier (var/spool/gammu/inbox) mais ensuite plus rien, rien n'apparait dans l'interface web. 

J'ai essayé de remonter le cheminement et les sms dans inbox ne s'inscrivent pas dans ma base SQL. J'ai fait un tour dans le script ParseSMS.sh et je ne vois rien d'anormal. Par contre, quand je le lance avec le terminal, il me renvoi (seq: opérande manquant), à ce moment je constate qu'un fichier .txt (Contenu vide) arrive dans mon dossier (var/spool/gammu/inbox) et disparait 60 secondes plus tard. 

J'en suis là !

Si quelqu'un à déjà rencontré ce problème où si vous avez une proposition de solution à m'apporter, je suis preneur. 

Merci par avance.

llimz, 2017-03-10:
Bonjour,

Dans ton fichier de configuration gammu-smsd se trouvant normalement dans /etc/gammu-smsd, as tu la ligne suivante

RunOnReceive = /var/www/html/RaspiSMS/parseSMS.sh

walmins, 2017-04-04:
j'avais le meme soucis et je me suis rendu compte que le problème venait des droits de mon dossier de réception des sms dans RaspiSms:
```echo "$sms" >> /srv/RaspiSMS/receiveds/"$date".txt```
pour faire des tests tu peux rediriger les sms vers ton dossier /tmp
```echo "$sms" >> /tmp/"$date".txt```
Merci

lp73, 2017-05-03:
Bonjour,
[RESOLU]
j'ai retrouvé ce problème en passant de la version packager à la version git master.
Je viens de trouver !
C'est la dernière ligne du fichier parseSMS.sh qui a été modifié :
original 
echo "$sms" >> /var/www/html/RaspiSMS/receiveds/"$date".txt
dernière version du master
echo "$sms" >> /var/www/RaspiSMS/receiveds/"$date".txt

Il faut indiquer le chemin du dossier receiveds

movvou, 2018-02-23:
Bonjour @walmins 
Chez moi tout semble ok pourtant les sms reçus ne s'affichent pas dans les conversations.

La dernière ligne de parseSMS.sh est bien conforme à mon arborescense (echo "$sms" >> /var/www/html/RaspiSMS/receiveds/"$date".txt)

La ligne "RunOnReceive = /var/www/html/RaspiSMS/parseSMS.sh" est bien présente dans /etc/gammu-smsdrc

J'en déduis que c'est peut-être une question de droits.

J'ai testé une redirection des sms vers /tmp avec "echo "$sms" >> /tmp/"$date".txt" et ça fonctionne.

Pour les droits actuellement, j'ai:
drwxr-xr-x  2 gammu    gammu    4096 févr. 23 01:14 receiveds
et pour les sms qui s'y trouvent
-rw-r--r-- 1 gammu gammu 31 févr. 23 01:01 20180223010126090542901.txt

SI vous avez reglé le problème en changeant les droits quelque part, vous pouvez peut être m'aider.

Merci d'avance,

closed 2018-02-27 by RaspbianFrance