RaspbianFrance/raspisms #72: Impossible de créer le SMS

john-75, 2017-02-07:
Bonjour,

J’essaie de mettre en place raspisms sur une debian et j'arrive à envoyer des SMS avec gammu en ligne de commande..
Par contre depuis l'interface web impossible j'ai le message d'erreur "Impossible de créer le SMS" et je ne vois pas d'où cela vient...
par contre je vois bien un sms reçu...

Merci par avance de votre aide...

romainguerrero, 2017-02-07:
Salut.
Il me semble que le problème peut arriver si jamais tu essayes de créer un sms dans le passé. Est-ce que le problème ne vient pas d'un problème de réglage de l'horloge de ton serveur ?
Est-ce que tu as le même message en essayant de le créer pour le lendemain ?

john-75, 2017-02-08:
J'ai synchronisé le serveur avec un serveur de temps (via ntp...)
J'ai testé l'envoi programmé d'un sms pour le lendemain et bien même erreur..
Est-ce que j'ai moyen de voir un log d'erreur quelque part ...
c'est vraiment un problème d'interface j'ai l'impression... le message ne va même pas en base mysql...
merci de ton aide

john-75, 2017-02-08:
Je viens de faire une réinstall complète en utilisant le fichier .deb et là tout est ok... un problème de dépendance peut être en effectuant l'install à la main...

closed 2017-02-08 by john-75