RaspbianFrance/raspisms #76: Support proxy

randomstuff, 2017-05-18:
I'm trying to use RaspiSMS behind a HTTPS nginx reverse proxy:

    server {
      listen 443 ssl;
      listen [::]:443 ssl;
      server_name example.com;
      # [...]
        location /RaspiSMS/ {
          proxy_pass http://backend;
        }
    }

The problem is that the generated URIs are in the form `http://backend/RaspisSMS/nyanya`. I could force the `Host` value but the URI scheme would remain `http`.

The relevant snippet in `constants.php`:

    define('HTTP_PWD', (isset($_SERVER['HTTPS']) ? 'https' : 'http') . '://' . (isset($_SERVER['SERVER_NAME']) ? $_SERVER['SERVER_NAME'] : 'localhost') . '/RaspiSMS/');

I'd be tempted to replace this with:

    define('HTTP_PWD', '//' . (isset($_SERVER['SERVER_NAME']) ? $_SERVER['SERVER_NAME'] : 'localhost') . '/RaspiSMS/');

which should be OK but would break the email here:

    $message  = "Votre compte a été créé sur le site " . HTTP_PWD . " avec les identifiants suivants:\n";

Another options would be to handle the `X-Forwarded-*` headers or the [RFC7239](https://tools.ietf.org/html/rfc7239) `Forwarded` header.

vincent1890, 2020-10-11:
Hello

I have exactly the same problem I am passing all my external accessible servers to HTTPS and the serv raspisms does not work in HTTPS how to fix this problem please?

Have a good day

closed 2021-02-14 by OsaAjani

randomstuff, 2021-02-14:
I'm wondering about the rationale about closing an issue because it's old. Is this issue not relevant anymore? Is this fixed somehow?

OsaAjani, 2021-02-14:
Hello, i've been closing issues that are too old and probably not relevant anymore, because the new version have evolve quite a lot internally. This solution may not be perfect, but it's in the objective of simplifying maintenance of the new version.

Originally i wanted to add this message to all closed issues but cannot find an easy way to do it in batch : 

> 
> Forwarding the release of RaspiSMS v3.0, internal functioning of RaspiSMS have been greatly modified, lot of fix have been implemented, and many features added.
> 
> This version change thefore resolved, directly or inderectly, the vast majority of currently opened "issues". Consequently, it have been decided to close all of the tickets.
> 
> However, if you think your "issue" is still relevant for the v3.0, you may open a new "issue", including a link to the previous one.
> 
>

If you think this issue is still relevant with v3.0, please open a new ticket with a link to the current issue.

randomstuff, 2021-02-14:
Ok thanks :)