RaspbianFrance/raspisms #77: ENVOIE DE SMS A LA VOLEE

MisterDGZ, 2017-05-29:
Bonjour
je voulais savoir s'il est possible d'envoyer de même SMS a plusieurs personne sans passe par les groupe
je m'explique
je voudrais faire un copier coller des destinataire par exemple 
06 66 66 77 88
06 88 99 66 77
07 88 88 55 22
je copie colle ces numéros et chacun reçois un sms

actuellement on doit les créer dans le carnet adresse et ensuite les attribuer a un groupe ce qui est très long et surtout notre liste de destinataire change a chaque fois

merci pour votre retour et votre outil super génial

Cordialement

David

MisterDGZ, 2017-05-29:
de plus si cela est possible je pourrais acheter une dizaine de raspberry pour chacun de mes collaborateurs

extrablind, 2017-06-21:
Vu comment RpiSms est conçu pour l'instant cela devrait demander du développement supplémentaire de la part des dévs.

closed 2021-02-14 by OsaAjani