RaspbianFrance/raspisms #78: To many massages?

meXxmasteR, 2017-06-20:
I have one contact he sends me 1000 sms but is loading and loading.
Can you help me or fix this?

thanks

extrablind, 2017-06-21:
To many "massage" is bad ;) 
Please precise your problem : you can't load conversation messages because there is too much, 
or it's spam and somebody sends you too much messages ?

meXxmasteR, 2017-06-21:
:P sorry for massage , i wrote this with stress @ work :(
Legitim wantes msg's from user jen (1000 sms received), when i will send its also loading and loading ..
Screenshot:

extrablind, 2017-06-21:
This is related to : #80. 
All messages are loaded in one time : due to lots of messages your script may fail. I think this is a defect and should be corrected.

closed 2021-02-14 by OsaAjani