RaspbianFrance/raspisms #79: Se passer du parseSMS.sh

extrablind, 2017-06-21:
Il est possible de lire une variable d'environnement avec PHP via la fonction getenv(). 
A mon sens le parseSMS.sh complique notre affaire puisqu'il rajoute une étape intermédiaire entre PHP et le démon Gammu. 

On pourrait appeler directement le script qui sert à parser les SMS depuis  la console php (internal console) dans la conf de gammu (runOnReceive). 
On économise une tache cron et un script bash. 

` # conf gammu :
runOnReceive= cd /var/www/raspisms/ &&  php -f ./console.php -c parseReceivedSMS
 `

Et il faudrait rajouter dans la commande parseReceivedSMS : 
`
<?php
 $content = "";
        for ($i = 1; $i <= getenv('SMS_MESSAGES'); $i++) {
            $number = getenv("SMS_{$i}_NUMBER");
            $content.= getenv("SMS_{$i}_TEXT");
        }
      # Save in Db
      # Do other stuffs...
`

RaspbianFrance, 2017-06-21:
Je pense qu'effectivement ce serait une super idée !
On planchera dessus pour la prochaine version !

meXxmasteR, 2017-06-21:
@extrablind nice idea, have you already tested and implemented in raspisms?
big thanks

extrablind, 2017-06-21:
This improvement is partially tested : I just run a php script which echo received SMS, number...etc and redirect the output to a file (see code up and add echo down the loop) : it works without problem.
I see no limitation to include that in RaspiSms. This is also an atomic solution : no need to parse entire received folder as soon as the script is execute for each sms received. To be clear : one sms one execution of the script with entire content rebuild and save after in the db.

closed 2021-02-14 by OsaAjani