RaspbianFrance/raspisms #80: Inverser l'ordre des messages dans les conversations et faire une requête ajax pour charger les messages plus anciens

extrablind, 2017-06-21:
Il parait plus logique que les derniers messages apparaissent en premier et qu'il ne faille pas scroller pour les atteindre. 
De plus, rien ne sert de charger inutilement les premiers message une requête ajax peut s'en charger à l'évènement scroll.

Florian

meXxmasteR, 2017-06-21:
Thanks im not alone :( have you an idea to fix this?
Idea: Not load alle message's load 20 - 50 and then make a facebook like loading - when scroll up then load more one's.

extrablind, 2017-06-21:
Yes, that's it : ajax requests on scroll and loading only firsts items could do the trick.

It's a priority enhancement in my opinion

jhillairet, 2018-03-11:
Une facon simple d'inverser l'ordre des messages (du plus récent au plus ancien), est de modifier la fonction `getmessages()` qui se trouve dans le fichier `templates/discussions/show.php`, en modifiant : 

`$.each(data.messages, function(key, message) {`
en
`$.each(data.messages.reverse(), function(key, message) {`

meXxmasteR, 2018-05-08:
Any fix the Problem or can give me tipps to fix it?

One User have 500 SMS in Discussion but its loading and loading.

closed 2021-02-14 by OsaAjani