RaspbianFrance/raspisms #85: Receive sms without numbers?

meXxmasteR, 2017-08-02:
Some sms like whatsapp or facebook or blablacar send not a number in the sms.
Facebook send facebook as sender "number". 
whatsapp send whatsapp as sender "number".
blablacar send blablacar as sender "number".

The Script can only parse numbers can you help me to parse the sms without sender number?

phoenix916, 2017-08-22:
hello it's may be interested for me too

any idear ?

meXxmasteR, 2017-08-30:
I have addet a "backup" line to the sh script.
SMS without Numbers NOT Display in the Webinterface but its echo'd in a backup Directory.

!!Create Directory smsbackup in the RaspiSMS Folder!!
Open:
/var/www/RaspiSMS/parseSMS.sh
On the End of the parseSMS.sh you found:
echo "$sms" >> /var/www/html/RaspiSMS/receiveds/"$date".txt
Edit it to:
echo "$sms" >> /var/www/html/RaspiSMS/smsbackup/"$date".txt
echo "$sms" >> /var/www/html/RaspiSMS/receiveds/"$date".txt

Voila your sms without Numbers will save in a directory smsbackup.

Demo:
BlaBlaCar:BlaBlaCar: Ihr BestÃ¤tigungscode lautet 49372.
FACEBOOK:Dein Facebook-BestÃ¤tigungscode lautet 1338959.

closed 2021-02-14 by OsaAjani