RaspbianFrance/raspisms #89: Erreur lors de l'envoi d'un SMS a un groupe...

pguillaumaud, 2017-11-10:
Bonjour,
D'abord, merci pour votre travail, raspisms est une excellente solution pour la gestion de SMS.

Je rencontre malgré tout un problème, quand je veut envoyer un sms a un groupe avec la commande suivante (par exemple):

`curl -X POST http://localhost/RaspiSMS/smsAPI/ -d 'email=xxx@yyy&password=xxx&groups[]=gginfo\&text=test SMS from horus pour le groupe gginfo'`

Rien ne se passe, et quand je vais voir les log apache, je lis ceci:

`[Fri Nov 10 12:32:13.519815 2017] [:error] [pid 19760] [client 172.19.50.250:44217] PHP Notice:  Undefined index: id in /var/www/html/RaspiSMS/controllers/smsAPI.php on line 95`

Bien évidement, le groupe 'gginfo' existe, et contient 2 contacts.
Une idée du problème?
Merci d'avance,
A vous lire,
Cordialement,

RaspbianFrance, 2017-11-10:
Salut, il me semble que ce problème a été réglé sur le github. Si tu
récupères la dernière version du fichier dans github ça devrait être bon.

Cordialement

pguillaumaud, 2017-11-13:
Re bonjour,
Effectivement, je me suis aperçu de ça en regardant les sources sur le GitHub, j'ai donc mis à jour ma version , mais j'ai toujours un problème, quand je veut envoyer a un groupe (qui existe et qui a 2 contacts) avec la commande suivante:
`curl -X POST http://localhost/RaspiSMS/smsAPI/ -d 'email=xxx&password=yyy&groups=gginfo\&text=[TEST] SMS pour le groupe gginfo'`
J'obtient la réponse:
`{"error":3}`
Manque de paramètres obligatoires ... lesquels ???
Merci d'avance de votre réponse,
Cordialement,

RaspbianFrance, 2017-11-14:
Ton adresse est bizarre, certain & ont un \ et d'autres pas, ça doit venir
de là !

pguillaumaud, 2017-11-16:
Bonjour
(bizarre, il me semblait avoir répondu à ta réponse .. enfin bref)
Effectivement, j'ai viré le '\' parasite, tout fonctionne impec, merci.

closed 2021-02-14 by OsaAjani