RaspbianFrance/raspisms #90: Pas plus de 10 contacts par groupe?

pguillaumaud, 2017-11-10:
Re bonjour,
On a créé un groupe qui doit contenir environ 80 contacts, mais à l'ajout des contacts dans le groupe, on est limité à 10 maxi...
C'est voulu? ou est-ce un paramètre que je n'ai pas vu?
Merci d'avance de vos conseils,
A vous lire,
Cordialement,

pguillaumaud, 2017-11-14:
Re bonjour,
Bon, j'ai trouvé le 'maxSelection' dans magicsuggest.min.js, je l'ai mis a 1024...

svinktux, 2018-01-19:
Bonjour, 
j'ai édité le fichier magicsuggest.min.js avec vim, reboot dans le doute le raspberry, mais pas mieux, je suis toujours bloqué a 10 dans le groupe déjà créé. 
une idée ?

daltoner, 2018-09-18:
Bonjour,
je viens d'installer un raspisms (en distri), j'ai le pb de limitation à 10 contacts, existe-t-il une solution ?
Merci

romainguerrero, 2018-09-18:
Bonjour,

Il faut effectivement rajouter dans les initialisations du plugin magicsuggest, le paramètre `maxSelection: null` pour désactiver totalement la limite ou `maxSelection: 80` pour l'augmenter à 80. C'est une des modifications apportées dans ma MR #38 dans laquelle vous pouvez retrouver les modifications à apporter dans les fichiers :

- templates/groups/add.php
- templates/groups/edit.php

Il est également possible d'augmenter le nombre de contacts ou de groupes à ajouter dans un sms en apportant la même modification dans les fichiers :

- templates/scheduleds/add.php (x2)
- templates/scheduleds/edit.php (x2)

Attention cependant, j'ai l'impression que si un sms contient trop de contacts, l'envoi n'est pas très bien géré mais je n'en suis pas certain.

ps : @GGSAS89 et @svinktux il ne faut jamais modifier manuellement le fichier d'une librairie mais simplement les paramètres qu'on lui passe au moment de son initialisation ;)

daltoner, 2018-09-18:
Merci, je suis impressionné par le délai de réponse ….

Génial.

Cordialement,
C. Danneville

[Nouvelle-signature-Cedric-Danneville]
[Signature-mail-ACTIVITE---PRO--RESEAUX]

daltoner, 2018-09-28:
Bonjour,

Tout d’abord encore merci de votre réactivité.
Je suis désolé, je ne vois pas comment modifier le paramètre max selection et l’update ne veux pas passer sur raspi.

Pourriez-vous m’expliquer plus en détail comment juste modifier le paramétrage pour pouvoir envoyer plus de 10 SMS ?

Merci d’avance.

Cordialement,
C. Danneville

[Nouvelle-signature-Cedric-Danneville]
[Signature-mail-ACTIVITE---PRO--RESEAUX]

closed 2021-02-14 by OsaAjani