RaspbianFrance/raspisms #92: Problème lors du lancement du portail

trist06, 2017-12-02:
Après l'installation de la version 2.0.1 sur une machine Debian 9 via le .deb ou le tar je rencontre le même problème

Not Found
The requested URL /RaspiSMS/connect/ was not found on this server.

pguillaumaud, 2017-12-04:
Bonjour,
Regarde les log apache, c'est peut-être un problème de path, la racine apache a changé depuis la dernière MàJ de raspisms, au lieu de '/var/www/RaspiSMS/...' il faut maintenant mettre '/var/www/html/RaspiSMS/...'.

Regarde aussi dans les fichiers de config, dans mvc/constants.php, vérifie les path, et jette aussi un oeil dans le fichier de conf du site (dans /etc/apache2/sites-available/raspisms.conf).

RaspbianFrance, 2017-12-04:
Salut, il manquait le fichier le fichier .htaccess dans l'installateur, il
a désormais été mis à jour, tu peux le récupérer via wget
https://raw.githubusercontent.com/RaspbianFrance/RaspiSMS/master/.htaccess

closed 2018-02-27 by RaspbianFrance