RaspbianFrance/raspisms #93: Fiabilité du Modem ...

pguillaumaud, 2017-12-04:
Bonjour,
Bon, j'ai des problèmes avec raspisms, certains SMS ne sont pas reçus.

Quand je vais voir l'affichage des sms envoyés, j'en vois certains qui ont la statut 'inconnu', et que l'on ne reçoit pas ...

Le modem est celui que vous préconisez dans votre doc, la clé USB UMTS HSDPA UMTS Huawei E169 Surfstick 7,2 Mbps (https://www.amazon.fr/HSDPA-Huawei-E169-Surfstick-contrat/dp/B004DEJEMY/ref=as_li_ss_tl?tag=raspbianfranc-21&_encoding=UTF8&psc=1&refRID=SC6KA0ZQ556YC4YWSC09&linkCode=ll1&linkId=1991ed2a3a85e6b629efcd006f151b92).

Comme j'avais déjà lu que certains utilisateurs ont rencontrés des soucis, j'ai programmé un reboot via cron le dimanche matin, ce reboot est opérationnel depuis l'installation du rapberry, mais j'ai quand même des pertes de SMS ...

Je suis pas un spécialiste telecom, mais comme le raspberry/modem est dans notre salle serveur, posé sur une étagère métallique dans une de nos baies, peut-être a-t'il un problème d'antenne en emission? dues a des parasites? ...

Je suis ouvert à toutes suggestions ....

jbd0101, 2017-12-04:
Bonjour ,
j'ai eu le même problème, est ce que le système marche au démarrage ou pas ?
as tu fais un usb modeswitch ?
je peux t assurer que le problème ne vient pas de l'endroit où il est mais que pour des raisons obscures , après un certain temps, l'usb est "rejeté" le meilleur moyen de savoir si l usb est monté (=> utilisable) est de regarder dans /dev (ls /dev ) si il y a des usb genre ttyUSB0, ttyUSB1,... Dans mon cas, la clé disparaissait et la solution était de de de faire une commande spéciale avec usb_modeswitch des que les logs disaient "device not exist" 

jc

pguillaumaud, 2017-12-04:
Bonjour,
Oui, le système (re)fonctionne après le reboot, j'ai mis en place un petit sccript dans /etc/rc.local qui m'envoi un SMS pour me signaler un reboot justement, et ce SMS je le reçois parfaitement.

Merci de l'info, je vais voir à logguer cette histoire de device USB, afin de voir ce qui se passe.

Tu a la syntaxe exacte de usb_modeswitch par hasard?

jbd0101, 2017-12-04:
oui pour la cle usb :  https://www.amazon.fr/Pumpkin-Adaptateur-7-2Mbps-Autoradio-Soutient/dp/B01G3GLPDU/ref=sr_1_2?s=electronics&ie=UTF8&qid=1512404054&sr=1-2&keywords=Huawei%20E169

il faut  : 
`sudo apt-get update
sudo apt-get dist-upgrade
sudo apt-get install ppp usb-modeswitch wvdial`
puis  editer /etc/usb_modeswitch.conf (le fichier est normalement vide de base )
`sudo nano /etc/usb_modeswitch.conf`
ajouter : 
`DefaultVendor=05c6
DefaultProduct=1000
MessageContent="5553424312345678000000000000061b000000020000000000000000000000"
Product: USB MMC Storage
Manufacturer: AnyDATA Corporation
SerialNumber: 000000000002
TargetVendor=05c6
TargetProduct=6000`
puis , à chaque démarrage il suffit de faire:
`sudo usb_modeswitch -c /etc/usb_modeswitch.conf`
sur ma cle, ca permet de faire en sorte qu elle soit lue comme une cle gammu sms non pas comme une cle de storage .
Il arrive que parfois ca "saute" et à ce moment là , il suffit de relancer la commande.
Je ne sais pas si c'est la même chose pour ta clé, mais en tout cas , il faut bien voir qu'il ait toujours des usb genre ttyUSB0.

JC

memento, 2017-12-11:
Bonsoir,
Avez-vous regardé cette [page](https://www.supinfo.com/articles/single/2892-mise-place-raspisms) de Supinfo qui relate quelques problèmes quand à la disparition du modème et propose des solutions ?

Bien à vous,
Memento

pguillaumaud, 2018-01-10:
Bonjour (et bonne année ...)
Bon, plus rien ne fonctionne depuis quelques semaines, aucun SMS n'est envoyé
J'ai tout essayé, modeswitch, reboot, débrancher/rebrancher, rien n'y fait ...
Tout est neuf, le système a à peine 1 mois, j'ose espérer que la clé Huawei n'est pas déjà tombée en panne ...
J'ai commandé un hub USB alimenté, en dernier recours, parce que le raspberry a clairement un déficit de courant sur les ports USB et ne peut pas tout alimenter, je verrais bien à ce moment la ...
Je vous tiens au courant.

ReaperSoon, 2018-01-16:
Bonjour, j’ai également des problèmes de déconnexion avec une clé gsm huawei sur un raspberry pi 3. Les sms partent et arrivent bien mais au bout de 1 jour environ plus rien. Je suis obligé de reboot ou débrancher/rebrancher ma clé pour qu’elle refonctionne (et envoie d’un coup tout les sms de regard)
Je vais tester vos solutions demain, merci

pguillaumaud, 2018-01-17:
Re bonjour,
Bon, j'ai installé un hub USB alimenté, branché au raspberry, la clé dans le HUB.
Reboot ... RIEN, plus AUCUN SMS ne part ...

un 'gammu identify' me dis:
"Aucune réponse dans le temps d'attente spécifié: Le téléphone n'est peut-être pas connecté."
Pourtant, tout est connecté...

Je relance gammu-smsd, ça me détecte le modem:
Périphérique       : /dev/ttyUSB0
Fabricant            : Huawei
Modèle              : E169 (E169)
Firmware             : 11.314.12.31.00
IMEI                 : xxx...
SIM IMSI             : yyy...

Mais toujours AUCUN SMS. J'ai beaucoup de msg d'erreurs dans les daemon.log, du genre:
Jan 17 15:23:23 raspberrypi gammu-smsd[1194]: New message to send: OUTC20180117_152319_00_+xxxxx_sms0.smsbackup
Jan 17 15:23:23 raspberrypi gammu-smsd[1194]: Message without SMSC, assuming you want to use the one from phone
Jan 17 15:23:23 raspberrypi gammu-smsd[1194]: SMS sent on device: "/dev/ttyUSB0" status=500, reference=-1
Jan 17 15:23:23 raspberrypi gammu-smsd[1194]: Error getting send status of message: Erreur inconnue. (UNKNOWN[27])

Au final, ça a bien marché pendant 2 semaines, et depuis, plus RIEN ... pas cool
Si vous avez des idées, je suis preneur ...

jbd0101, 2018-01-17:
j ai eu ce problème , je l'ai "corrigé" grâce à un fichier cron qui regarde si la cle est prénte ou non , dans mon cas, si elle ne l'est pas il m'envoit un message slack , tente de la "remonter" comme une clé modem et si il n y arrive pas il reboot .
C est pas top comme programme mais ça marche plutot bien : 
`

function post_to_slack () {
  # format message as a code block ```${msg}```
  SLACK_MESSAGE="$1"
  SLACK_URL=https://hooks.slack.com/services/xxxxxxxx/xxxxxxxxxx
  curl -X POST --data "payload={\"text\": \"${SLACK_MESSAGE}\"}" ${SLACK_URL}
}

gammuLog="/var/log/syslog"
myLog="/var/log/perso/mobile-control"
myPhoneNumber="048555555555"

nErrors=$(tail -n 12 /var/log/syslog | grep -E "DEVICENOTEXIST|TIMEOUT|DEVICEOPENERROR|DEVICELOCKED|DEVICEWRITEERROR|DEVICEREADERROR" | wc -l)
if [[ $nErrors -gt 3 ]]; then
    gammu-smsd-inject TEXT "$myPhoneNumber" -text "Rebooting because of too many errors"   # may not work, as the connection to the device is lost
	#echo "[resetusb-workaround] Rebooting because of too many DEVICENOTEXIST" >> "$myLog"
	logger -s "Je voudrais rebooter erreur usb"
	logger -s "reinitialisation de l usb"
	sudo usb_modeswitch -c /etc/usb_modeswitch.conf
	sleep 20
	logger -s "je regarde si ca marche "
	post_to_slack "usb non initialise, je tente de la reparer"
	
	nErrors=$(ls /dev | grep ttyUSB0 | wc -l)
	if [[ $nErrors -eq 0 ]]; then
		post_to_slack "*ATTENTION A TOUT LE MONDE* :J AI UN PROBLÈME : PLUS DE CONNEXION USB , VEUILLEZ ME REBOOTER AU PLUS VITE "

		logger -s "je REBOOT"
	fi
	sleep 15
	sudo reboot
fi

`

ReaperSoon, 2018-01-17:
En effet cette solution n’est pas viable dans certains cas par exemple dans mon cas où le Raspberry est utilisé pour d’autres fonctionnalités qui nécessite de ne pas reboot (Kodi dans mon cas)

pguillaumaud, 2018-01-18:
Bonjour,
Le reboot, j'en fait déjà un une fois par semaine, je force un reboot par cron le dimanche, jusqu'à présent ça fonctionnait, mais plus depuis ces derniers temps ...
J'ai tout essayé, usb_modeswitch, débrancher/rebrancher, reboot, lier le device à un nom générique pour éviter le changement de device USB, mais plus RIEN ...

Je vais tester le modem sur un PC windows, pour voir si c'est pas lui qui déconne (si oui, ça le fait moyen un truc neuf qui dure moins d'un mois ...).

Après ça, à part tout réinstaller, je vois pas...

closed 2021-02-14 by OsaAjani