RaspbianFrance/raspisms #94: Problèmes d'envoi d'un sms

Nems27, 2017-12-15:
Salut ,

Je rencontre un problème concernant l'envoi d'un sms
😦 … Lorsque je lance cette commande dans la console ' curl -g http://sms/RaspiSMS/smsAPI/?email=XXX@XXX.fr\&password=XXXXX\&numbers=06XXXXXXXX\&contacts=toto\&groups[]=groupe\&groups[]=test\&text=TestMultiGroupe ' Seul “numbers” reçoit le message mais le contact et les groupes non… Avez-vous des idées?

Merci d’avance de votre aide.

pguillaumaud, 2017-12-15:
Bonjour,
J'ai rencontré le même souci, la réponse est la:
https://github.com/RaspbianFrance/RaspiSMS/issues/89

Nems27, 2017-12-15:
@pguillaumaud D'accord merci beaucoup je vais regarder ca

Nems27, 2017-12-18:
Merci beaucoup @pguillaumaud , pour les futurs personnes qui auront  ce problème, vous pouvez simplement récupérer le fichier smsAPI.php

Blamo27, 2017-12-21:
@Nems27 Merci

closed 2017-12-30 by Nems27