RaspbianFrance/raspisms #98: Received SMS do not appear in RaspiSMS dashboard

jhillairet, 2018-02-24:
Bonjour,

Tout d'abord merci pour votre travail !

J'ai installé RaspiSMS à priori avec succès : j'arrive à envoyer des SMS et les SMS envoyés apparaissent dans le dashboard.

Par contre, les SMS recus arrivent bien via gammu (`/var/spool/gammu/inbox`), ils sont copiés dans le répertoire `received` de RaspiSMS, mais par contre rien n'est envoyé dans la base de donnée (`select * from receiveds` renvoie un empty set). Comment puis-je débugger un peu plus?

jhillairet, 2018-03-03:
Solved with PR #99

closed 2018-03-03 by jhillairet

domaze, 2018-03-10:
Merci pour ce travail. En ce qui me concerne ça a fonctionné grâce à cette modif  : 
dominique@raspberrypi:~ $ sudo /var/www/html/RaspiSMS/console.php -c parseReceivedSMS
Analyse du SMS 20180309221342078259357.txt
FROM [Controller : 5aa3a325d412b - internalConsole] - We found 1 users
FROM [Controller : 5aa3a325d412b - internalConsole] - Password is valid
FROM [Controller : 5aa3a325d412b - internalConsole] - We found 1 commands
FROM [Controller : 5aa3a325d412b - internalConsole] - We found command TEST1
FROM [Controller : 5aa3a325d412b - internalConsole] - And the count is ok
Execution de la commande : TEST1 :: /var/www/html/RaspiSMS/scripts/cd1.sh

Hatim27, 2018-07-11:
Merci pour ce super bijou, tout fonctionne bien à part que les sms reçu ne s'envoient pas vers la base ce qui fait que je les vois pas sur l'interface..
J'ai essayé la solution décrite par jhillairet mais sans résultat.

LaurentThiol, 2018-07-16:
Logique - erreur de frappe c'est :
`if ($dir == '.' || $dir == '..' || $dir == '.to-keep')`

triseaux, 2018-10-03:
Bonsoir,

Deja merci pour les infos et le travail effectuer.

J'ai fait ma modification du internalConsole.php mais j'ai toujours un souci de lecture des sms reçus.

` sudo /var/www/html/RaspiSMS/console.php -c parseReceivedSMS
Analyse du SMS .tokeep
FROM [Controller : 5bb5193d3a345 - internalConsole] - Unable to read file ".tokeep
`

Une idée ?