RaspbianFrance/raspisms #101: Raspberry (raspbian stretch) raspisms install errors

zicmama, 2018-03-21:
Je viens de lancer l'installation du paquet raspisms-2.0.1.deb

Il y a une erreur lors de la création de la base MySQL:

> Le fichier de configuration /etc/gammu-smsdrcrc n'a pas pu être généré. Vous devrez le configurer à la main.
> La base de données de RaspiSMS va être déployée.
> Création de l'utilisateur MySQL ''raspisms''
> raspisms password = password
> L'utilisateur MySQL 'raspisms' a bien été créé. Vous pouvez retrouver son mot de passe dans le fichier '/var/www/html/RaspiSMS/mvc/constants.php'.
> ERROR 1071 (42000) at line 123: Specified key was too long; max key length is 767 bytes
>  Une erreur est survenue lors de la création de la base de données, effectuez cette création à la main avec la commande 'mysql -u raspisms -ppassword < /var/www/html/RaspiSMS/createDatabase.sql'
>

jhillairet, 2018-03-21:
Il y'avait une erreur dans le fichier de définition de la base de donnée : cf le commit e585647583a49036bf00fd60983cd20259186a8a. Il faut corriger la valeur 255 en 191 et réessayer.

zicmama, 2018-03-21:
J'ai téléchargé le deb ici
https://raspisms.raspbian-france.fr/download/raspisms-2.0.1.deb

zicmama, 2018-03-22:
Après correction de la creation de la base MySQL, quand j'accède à RaspiSMS, je me trouve confronté à l'erreur.

Erreur : SQLSTATE[HY000] [1698] Access denied for user 'root'@'localhost'

zicmama, 2018-03-22:
Dans "/var/www/html/RaspiSMS/mvc/constants.php", il faut changer le user (par défaut root / sans mot de passe) en "raspisms / password"

Le scprit de confiugation informe de ce paramètre à l'installation (création de la base MySQL), mais le fichier constants.php n'est pas conforme.. 

```
La base de données de RaspiSMS va être déployée.
Création de l'utilisateur MySQL ''raspisms''
raspisms password = password
L'utilisateur MySQL 'raspisms' a bien été créé. Vous pouvez retrouver son mot de passe dans le fichier '/var/www/html/RaspiSMS/mvc/constants.php'.

```

jhillairet, 2018-03-23:
Effectivement, j'ai eu le même souci, c'est bien de le documenter (ce que je n'ai pas fait).  

Attention : si on lance plusieurs fois de suite le script d'installation, cela accumule les lignes dans le crontab. Il faut alors enlever les doublons dans le crontab (`sudo crontab -e`) sinon les SMS seront envoyés en double ou plus.

closed 2021-02-14 by OsaAjani