RaspbianFrance/raspisms #102: Login impossible RaspiSMS

Dowz95, 2018-04-04:
Bonjour, je suis actuellement en train de réaliser un projet pour mon Bac qui est liée avec la création d’un site internet via un Raspberry Pi 3, j’ai réussie toutes les étapes pour créer un serveur web mais viens le moment où j’ai du installer le module RaspiSMS et l’intégrer au site web, j’ai suivi plusieurs tuto trouvés sur le forum Rasbian France ou autre et au final l’installation c’est dérouler correctement, j’ai accès à la page de connexion à partir de http://Mon-ip/RaspiSMS mais impossible de me connecter avec les identifiants définies durant l’installation, j’ai bien édité le fichier conn_bdd.php situé dans /var/www/html/RaspiSMS/mvc pour relié la base de données MySql, si les données n’étaient pas correctes je n’aurais même pas accès à la page d’accueil, je pense qu’il s’agit d’un problème de droit d’accès pour la lecture de la base de données raspisms créer avec le fichier createDatabase.sql, j’ai bien vérifié si dans la base de données la table “users” contenais mes identifiants et ils y sont bien, je ne sais pas donc quoi faire pour régler mon problème.

jhillairet, 2018-04-06:
Bonjour,

As tu regardé les bugs soumis précédents (et résolus?). Notamment
concernant le script d'installation ?

Sinon, puisque tu as accès à la base, tu peux soit créer un nouvel
utilisateur soit modifier le mot de passe de celui que tu as créé ?

Dowz95, 2018-04-06:
Qu'entend-tu par script d'installation ? 
J'ai suivi plusieurs tutorielles pour installer RaspiSMS et l'installation c'est déroulée correctement.
Concernant la base de donnée j'ai essayé de créer d'autre utilisateur, de modifier le mot de passe mais rien à faire, toujours le même problème

OsaAjani, 2021-02-14:
Suite à la sortie de la version 3.0 de RaspiSMS le fonctionnement interne de RaspiSMS a été largement modifié, bon nombre de fix pour des bugs mineurs ont été intégrés, et plusieures fonctionnalitées ajoutées.

Ce changement de version a donc résolu, directement ou indirectement, la grande majorité des "issues" actuellement ouvertes. Il a donc été décidé de fermer l'ensemble des tickets.

Si toutefois vous pensez que votre "issue" est toujours pertinente pour la version 3.0, vous pouvez ouvrir une nouvelle "issue", en y incluant un lien vers l'ancienne.

-----------------

Forwarding the release of RaspiSMS v3.0, internal functioning of RaspiSMS have been greatly modified, lot of fix have been implemented, and many features added.

This version change thefore resolved, directly or inderectly, the vast majority of currently opened "issues". Consequently, it have been decided to close all of the tickets.

However, if you think your "issue" is still relevant for the v3.0, you may open a new "issue", including a link to the previous one.

closed 2021-02-14 by OsaAjani