RaspbianFrance/raspisms #103: Statut sms inconnu

tartreau, 2018-04-13:
Bonjour, j'ai développé un micro interface web en parallèle de celle de RaspiSMS pour certaines raisons. Elle est fonctionnelle, les SMS s'envoient correctement. Or quand je vais sur l'interface pour voir le statut du sms que j'ai envoyé, le statut est "inconnu" alors que j'ai bien reçu le sms. 
Je me suis dit que ça venait de mon interface, mais même problème avec l'interface de RaspiSMS. 
J'enregistrer les numéros au format 0XXXXXXXXX, est-ce que cela peut venir de là ?

RaspbianFrance, 2018-04-13:
Salut, c'est possible que ça ai un impact en effet, essaye au format +33 et
regarde si ça corrige le problème.

tartreau, 2018-04-14:
J'ai essayé avec un +33 mais le SMS ne s'est pas envoyé, à quoi doit ressembler la nouvelle URL ?

tartreau, 2018-04-16:
bonjour, en remplaçant un 06 par un +336 ça n'envoie pas de SMS mais j'ai bien le code error 0

tartreau, 2018-04-30:
(up de l'issue)

thomasbonsirven, 2018-06-04:
Tu envoie tes sms avec la nouvelle interface ou celle du Raspisms ?

tartreau, 2018-06-04:
J'ai essayé avec les deux (interface de base et celle de raspi) et avec les numéros en +33 et en 06, même résultat pour tout (statut inconnu)

thomasbonsirven, 2018-06-05:
Et on est d'accord que tu ne receptionne pas le sms ? As tu testé de faire un teste en direct ? en ligne de commande avec gammu ?

tartreau, 2018-06-06:
si je réceptionnes, j'ai juste pas de statut
pas testé avec gammu

thomasbonsirven, 2018-06-07:
Ok peut tu lancer le .deb de raspisms correspondant à ta version ? Je me demande si ce n'est pas un pb de script dans raspisms.

attention, cela va te doubler les taches cron !

Ulrick17, 2020-07-16:
Bonjour 

Nous avons refait le .deb et on a virer les doublons dans le crontab mais nous avons toujours le meme soucis.

Les messages sont bien envoyés et reçus mais le statut dans l'interface reste à unknown

Merci de votre retour

Ulrick17, 2020-08-04:
Bonjour 

Est ce que quelqu'un a une solution pour avoir le statut au niveau de l'interface

TiCaillou, 2020-08-25:
Bonjour, j'ai le même problème. J'arrive à envoyer des sms en ligne de commande `gammu sendsms TEXT 06xxxxxxxx  -text "Hello"` Réception du sms OK. Par l'interface RaspiSMS, j'envoi ne se fait pas et la colonne statut est à "inconnu".
Cela peut-il venir d'une mise à jour de gammu ? je suis en 1.40

Ulrick17, 2020-08-25:
Je suis absent(e) du bureau jusqu'au 31/08/2020

Merci d'envoyer le mail à technique@dii.fr

Remarque : ceci est une réponse automatique à votre message  "Re:
[RaspbianFrance/raspisms] Statut sms inconnu (#103)" envoyé le 25/08/2020
09:04:32.

C'est la seule notification que vous recevrez pendant l'absence de cette
personne.

closed 2021-02-14 by OsaAjani

PacifickFR42, 2023-09-12:
UP.
Bonjour, j'ai le même problème !