RaspbianFrance/raspisms #104: Ajouter une commande (script)

thomasbonsirven, 2018-04-14:
Bonjour à tous,

Bon, déjà raspisms est vraiment efficace. 

Alors ma demande est :
Comment je peux automatiser un transfert de SMS.
Exemple : j'aimerais envoyer un SMS comme ceci :

Groupe:1
Coucou à tous

Le raspisms le reçois et le traité comme suit :
Groupe:1 = le groupe de contact joueurscs ( dont 1 est l'id)
Il y a dans ce groupe 25contacts
Il envoie 25 SMS dont le message est :

Coucou à tous

Est ce possible ?
Je suis avec raspisms2.0

thomasbonsirven, 2018-04-18:
Bonjour,

Personne pour m'aider ?

thomasbonsirven, 2018-04-24:
Bonjour

Je vois que ce n'est pas très actif. Cela est réellement dommage. J'aimerais bien contribuer à ce projet.

J'ai compris pas mal de choses mais toujours pas pour la création de commande....

Un personne charitable pour expliquer, s'il vous plaît.

thomasbonsirven, 2018-04-24:
Bonjour et merci,

Ok peut on avoir un exemple ? 

Dans le menu "commande" il sert à quoi ?

thomasbonsirven, 2018-04-24:
Enfaite en réfléchissant il faudrait avoir une liste de commande type disponible qu'on peut faire, cela serais super.

Je présume que le menu "commande" sert à cela.

Donc à l'intérieur d'un .sh on devrait voir des commande de type gammu ???

RaspbianFrance, 2018-04-26:
En fait la partie commande sert à créer des commandes personnalisées
permettant de déclencher ds scripts hébergés sur le serveur.
Seulement pour le moment le déclenchement de ces commandes demande de
respecter un format très précis, ce qui ne correspond pas à un usage pour
des utilisateurs grand public.

Pour plus d'informations tu devrais regarder sur la documentation de
RaspiSMS.

Je souhaiterais ajouter sur la prochaine version un moyen de faire de
l'analyse directement sur les messages, genre chatbot un peu limite, mais
ce n'est pas pour tout de suite.

thomasbonsirven, 2018-04-28:
Hello,

Bon je commence à comprendre le fonctionnement.

Le formatage du SMS ne me pose pas plus de problème, si ce n'est les MDP dans le SMS.

Par contre impossible de faire un déclanchement d'une commande. J'ai suivi l'exemple, avec un script qui envoie une réponse "ok bien reçus" à la personne qui a envoyé un sms.

Le formatage est bien respecté : [test][login:(mon login)][password:(monmdp)].

La commande créé dans raspisms est bien test. Il réceptionne le SMS, efface le MDP mais dans les logs il ne signale pas que la commande a été déclanché : commande = non.

Peut tu m'aider ?

RaspbianFrance, 2018-05-02:
Salut,
tu devrais commencer par un script plus simple, genre qui fait juste un
touch /home/pi/toto.test pour voir si le fichier est créé. Par ailleurs, il
faut un ":" après ton "test" il me semble.

Pozinux, 2020-01-15:
Bonjour, je rencontre le même soucis. Je creuse. En tous cas merci pour ce superbe outils. :)

closed 2021-02-14 by OsaAjani