RaspbianFrance/raspisms #107: Transfert des SMS de réponse vers un mail

thomasbonsirven, 2018-04-24:
Bonjour

Je cherche le moyen de transférer les SMS de réponse vers le mail d'une personne ou de la personne qui a envoyé le mail.

Ssmtp est installé.

Je reçois tous les mails de la raspberry. Des mes autres mini-sites, mais pas des SMS de réponse.
Malgré le transfert de SMS activé et le mail renseigné.

À oui, je reçois bien le mail quand on créer un compte.

Comment faire ?

Merci d'avance.

thomasbonsirven, 2018-05-06:
bonjour

pas de nouvelles pour cette demande ?

thomasbonsirven, 2018-06-05:
Up up !! :)

sakar33, 2020-08-31:
Bonjour, 

si ssmtp est bien installé et configuré, j'ai réussi à faire le transfert de mail en ajoutant une ligne à la crontab

FredMa01, 2020-10-31:
Le transfert de mail fonctionne bien mais il semble y avoir un souci d'encodage :
```
Le numÃ©ro +33xxxxxxxxxx vous a envoyÃ© un SMS : 
C'est un essai plutÃ´t rÃ©ussi
```
Une idée pour résoudre le problème ? Le souci vient-il de raspisms ou d'exim4 ?
Merci
Fred

FredMa01, 2020-10-31:
Le problème ne vient pas de raspisms mais bien d'exim4 :
Lorsque je lance la commande :
echo "Là, c'est réjouissant !" | mail -s "C'est l'été" monadresse@exemple.com
Tous les accents du mail reçu sont remplacés par des caractères spéciaux.

Tout va mieux si j'ajoute l'header : “Content-Type: text/plain; charset=UTF-8” :
echo "Là, c'est réjouissant !" | mail -a "Content-Type: text/plain; charset=UTF-8" -s "C'est l'été" monadresse@exemple.com

Il faut donc que je cherche comment forcer cet ajout d'header UTF-8....

closed 2021-02-14 by OsaAjani