RaspbianFrance/raspisms #108: Changer expéditeur

tartreau, 2018-04-24:
Bonjour, je voudrais savoir si dans RaspiSMS il était possible d'afficher un nom lors de la réception du SMS plutôt que le numéro. 
Si oui ou faire le changement ? Dans RaspiSMS ou il faut fouiller dans des fichiers ?

thomasbonsirven, 2018-04-28:
Bonjour

Effectivement, je cherche aussi la solution.
Il n'y a pas de possibilité dans linterface. ( Du moins de ce que j'ai vu ).

Je cherche donc dans les fichiers.

Je te tiens au jus.

RaspbianFrance, 2018-04-29:
Salut, non désolé ce n'est pas possible.
En fait le changement de nom lors de l'affichage du numéro est une sorte de
trucage qui est effectué directement par les opérateurs téléphoniques.
La seule solution serait donc de trouver une carte sim chez un opérateur
offrant cette possibilité.

tartreau, 2018-04-30:
D'accord, est-ce que vous connaissez la procédure à suivre pour cela ?

RaspbianFrance, 2018-05-02:
Salut, désolé non pas du tout.
Comme c'est géré par l'opérateur, je présume qu'il faudrait leur demander
directement. Mais il doit s'agir de contrat spéciaux de type commerciaux.
Le plus simple serait peut-être de contacter le support des grands
opérateurs (par twitter par exemple) pour leur demander.

tartreau, 2018-05-02:
Très bien merci !

thomasbonsirven, 2018-06-04:
Bonjour,

Pour information, 

Free ne fait pas ce genre de service.

Il y a Orange devis à 99€ht par mois ( tres chere )
SFR avec un devis à 79€/mois ht
Bouygues Pro ( 49€/mois TTC ) 
Entreprise (79€/mois HT )

Services only sms sans aucune limitation, avec négociation ( par confirmé pour Bouygues Pro )

Il en ressort que cela reste un service pour les entreprises, ou association ;)

closed 2021-02-14 by OsaAjani