RaspbianFrance/raspisms #109: Multi Sim

tartreau, 2018-05-09:
Bonjour, 
Je voudrais savoir s'il était possible d'installer plusieurs cartes SIM sur le RaspiSMS mais que cela garde les mêmes contacts et tout ce qui vas avec.

Ma carte Sim est connecté au Raspi via une clé USB. 
J'ai vu cette issue : https://github.com/RaspbianFrance/RaspiSMS/issues/49
Elle date d'il y a deux ans et dit que ce n'est pas trop possible en raison des surtensions qu'il pourrait y avoir. Est-ce encore d'actualité aujourd'hui ?

thomasbonsirven, 2018-06-04:
Hello,

j'ai fait un test, cela fonctionne plutôt bien.
Mais tu doit faire des modifications, coté raspisms et Gammu, selon moi, et selon les tests effectué. 

J'ai fait une sorte de "Load Balancing" entre mes sim's :)

Je m'explique :
Raspisms fait appel a gammu avec le fichier de configuration gammu se trouvant dans /etc/gammurc.
En m'inspirant de la documentation de Gammu : https://fr.wammu.eu/docs/manual/config/#working-with-multiple-phones, j'ai créer un petit sh qui permet d'envoyer pleins de sms simultanément repartie en 2.

En d'autre terme, quand j'ai 6 sms je repartie en 2 listes :
- Sim 1 pour les 3 premiers
- Sim 2 pour les 3 derniers
Si j'en ai 5 : 
- Sim 1 pour les 3 premiers
- Sim 2 pour les 2 derniers

Coté PI 2 : 

Pour le cote "électrique" et "alimentation" du Raspi, je suis avec un chargeur 5v 2.5Amp.
Ma consommation est de 1.697amp avec un modem + 0.374amp avec le 2eme.
Le Raspberry fonctionne bien, RAS dans un test a court terme.

Coté PI 3 : 

Alors effectivement, j'ai plus de mal avec le 2eme modem, mais j'ai aussi un ecran 7". Avec une alimentation de 2.5amp ( 5v ) le deuxieme modem s'allume puis s'eteint, et continue comme ca.

J'ai débrancher mon écran ( alimenté via le GPIO de la PI ) et la cela fonctionne.

Ma conclusion : 
Je te conseil d'avoir une très bonne alimentation ( chargeur ) de bonne manufacture et limiter les usage avec la pi pour sms. Ne pas trop la charger avec des processus interminable demandant du GPIO + USB + HDMI etc ...
Les surtension viennent principalement des chargeurs ( bloc alimentation ). Cela n'est pas très commun de nos jours que çà proviennent des cartes de type Raspberry. A moins d'avoir de mauvais branchement.

Éventuellement, regarde a mettre tes usb via GPIO, cela permet un frottement limité entre les connexions, ce qui fait aussi des surtension.

en éspérant avoir répondu a ta demande de départ :)

Nico80, 2018-10-29:
Bonsoir, 
Je souhaiterai gérer l'envoi par plusieurs carte sim depuis mon raspberry (avec deux clés USB).
Malheureusement après plusieurs heures de test et configuration, les sms partent toujours avec la même carte sim en fonction de celle qui est branchée en premier...
j'ai testé des modifications dans gammu-smsdrc mais cela ne change pas la carte sim 
j'ai essayé le tuto : https://back2arie.wordpress.com/2010/07/27/using-gammu-smsd-with-multiple-phone/

Je reste sans solution... 
Si quelqu'un a une idée?? 
thomasbonsirven  comment as-tu fait?

Merci, beaucoup.

closed 2021-02-14 by OsaAjani