RaspbianFrance/raspisms #111: Raspisms lenteur / envoie d'un sms

thomasbonsirven, 2018-06-03:
Bonjour,

bon, je suis embenté, l'envoie de sms est tres lent.
Exemple, j'ai envoyé plus de 100 sms, cela a pris 3h30 environ....

J'ai fait un test, 1 sms envoyé via raspisms prend environ 25 à 35 secondes.
Avec la commande en ssh : 
gammu-smsd-inject TEXT 0612345678 -text "testing 001 ...002... "
j'ai recus le sms en 7 secondes environ.

Comment faire ? Qu'est ce que je peux faire ?

Merci d'avance.

thomasbonsirven, 2018-06-07:
Pour info,
j'ai changé la config gammurc : 

```
[gammu]
device = /dev/ttyUSB0
name = Modem
connection = at115200

[gammu1]
device = /dev/ttyUSB1
name =  modem
connection = at115200
```

Et j'ai mis a jours Raspisms en version 2.0.1 avec php7.0.24
Résultat : 40 sms en 30min. Mais je trouve cela encore tres long.

Une idée ?

LaurentThiol, 2018-07-16:
1. Quel est ton modem ? Vériifer la config dans : [https://wammu.eu/phones/](url)

2. Il est préférable de fixer le chemin port USB par cette méthode [http://blogmotion.fr/diy/tutoriel-gammu-cle-3g-dongle-16409](url)

3. Verifie avec uptime et top si il n'y a pas un service parasite sur ta config

closed 2021-02-14 by OsaAjani